ArduPilot/ardupilot #1721: arming error if a enabled sensor is not found

tridge, 2015-01-01:


gmorph, 2015-01-10:
Need to add a check that if they have configured an  airspeed sensor but it didn't find one.   The same for Rangefinder and compass.
Need to check that a message appears on the console when arming fails which clearly indicates what has failed.
Also need to check that the LED flashes and buzzer sounds in a consistent way that copter handles this.
Needs to check if you have more then one type of the same sensor they are reporting - so multiple IMUs (Pixhawk) do a check - do in the same way Copter handles this.

tridge, 2015-01-20:
done for airspeed and compass
not done for rangefinder yet

magicrub, 2016-06-06:
done for GPS2

rmackay9, 2019-01-05:
Still not done for range finders yet I'm afraid.

@Naterater, Thanks for the big issue list review by the way!

peterbarker, 2020-06-18:
@WickedShell did rangefinders.  All done.

closed 2020-06-18 by peterbarker