ArduPilot/ardupilot #1724: L1 Guidance in plane has track offset when roll angle estimate is offset

priseborough, 2015-01-02:
Steady state errors in the estimated roll angle cause a cross track error. Adding some form of integrator in the L1 guidance to compensate for this type of error should be investigated.

iskess, 2015-01-02:
Is the is this related to the airplane tracking right of course when in a sideslip? There are a few threads on the matter: http://ardupilot.com/forum/viewtopic.php?f=98&t=7913&start=10

priseborough, 2015-01-02:
Side-slipping is one source of error, a difference between estimated bank and true bank angle is another, so yes they are related in the sense than both cause a cross-track acceleration that biases the trajectory.

The plane does correct via bank angle, but the amount of bank angle correction is proportional to the error. I have resisted adding an integrator term due to the complexities associated with ensuring it handles transient errors during track capture and turns and because there is already adequate adjustment to remove both side-slip and AHRS errors.

If your plane isn't side-slipping, has a track offset and the measured roll tracks the demanded roll, then it is likely it thinks it is flying straight and level when it is in fact banked slightly. 
With sticks centered in FBWA mode it should fly wings level in a straight line without side-slipping. You can adjust AHRS_TRIM_X so that that when the wings are leveled, the roll angle from the AHRS is zero. If you do this and it is not side-slipping, then it should not have a cross-track error.

It sounds like you need to trim it in manual to fly straight and level without side-slip, repeat the RC calibration so that in auto modes it is using the same trims, make sure you are using some RLL2SRV_I and  then adjust AHRS_TRIM_X so that it flies straight and level in FBWA.

priseborough, 2017-06-21:
This was fixed with addition of a cross track error integrator gain to the L1 controller.

closed 2017-06-21 by priseborough