ArduPilot/ardupilot #1745: Copter:Pitch automatically raised after arming in ArduCopter 3.2stable.

touma-setsuna, 2015-01-07:
After Pixhawk is armed,HUD shows the pitch raises automatically, for about 5 - 10 degree. From  power on to this problem occurred, the board haven't been touched or moved.(powered by PM). Two screenshots may show this problem clearly. 

[image]
Before arming, the HUD shows the board is level.

[image]

After armed, the HUD shows the nose of board is pitching up slowly, then stops in the postion of about 5-10 degree up. 

During this, the board stays level itseif. 

And this WON'T occur in firmware 3.1.5.  A friend of mine finds this bug as well.

R-Lefebvre, 2015-01-07:
Which AHRS system are you using, DCM or EKF?

touma-setsuna, 2015-01-07:
DCM. I'm trying to upload my log files but failed many times. Great Firewall may be the cause.

R-Lefebvre, 2015-01-07:
Jonathan, do you have any idea on this one?

rmackay9, 2015-01-08:
This will likely be related to two things:
1. the difference in the accelerometer weighting between armed and disarmed (the accels are more heavily weighted when disarmed)
2. the gyro biases are recalculated at the first arming.
The shift in attitude will be because the gyro biases are pulling the attitude solution in one direction while the accels are pulling in the other.  when the accels become less weighted the solution moves towards the gyros.

Why the gyro biases are off is probably the underlying question.  My guess is if you leave the vehicle for a few minutes and arm again the attitude won't shift.

closed 2018-01-12 by lthall