ArduPilot/ardupilot #1746: Copter - Log all mission commands/waypoints to dataflash when vehicle arms

JoshWelsh, 2015-01-07:
For troubleshooting purposes it would be helpful to know what the mission is that a copter has programmed when it arms.  This will help streamline post-analysis of logs for data examination as well as user support and remove the need to ask the question, "can you send me the mission as well?"

I would say this is a low priority, nice to have thing, but helpful nonetheless.

touma-setsuna, 2015-01-07:
The Great Firewall prevented me from uploading everything. I've tried 10 times to upload 2 images. Looking for Stable&Free “ladder”......

rrr6399, 2015-01-07:
Logging the initial mission definition is definitely needed for troubleshooting. Any mission updates during flight should also be logged as well.

R-Lefebvre, 2015-01-07:
I'll see if I can do this with my other logging changes I'm working on.

rmackay9, 2015-09-09:
Tom Pettinger has added this to AC3.3.

closed 2015-09-09 by rmackay9