ArduPilot/ardupilot #1748: Copter: MAVlink-only arming

R-Lefebvre, 2015-01-07:
Currently, it is not possible to arm a copter via MAVlink (tablet) without a Tx, unless you set either Arming_Check to 0, or FS_Throttle to 0.  Both of these options present significant risks to use.  A method should be developed to allow MAVlink arming while bypassing RC-Input throttle checks.

silviorevelli, 2015-05-09:
I also experienced that problem ("Throttle Too High"), but only with SITL. With a real copter you just set rc 3 to 1100 prior to arming, so you shouldn't need a TX (tested). Please take care to center also all the other channels if you're flying without TX, otherwise you'll experience a "beautiful" takeoff...

Pedals2Paddles, 2020-02-06:
Closed due to no longer an issue. There is an `arming_check` bit for RC now.

closed 2020-02-06 by Pedals2Paddles

faerietree, 2020-07-25:
@Pedals2Paddles Matt, is this the reason why the MAV link command way to set do_pre_arm_checks parameter to `arm` method has been removed? Could you help me to find where to set the bit you mention in MAVproxy? Do you know of any example? Thank you

https://github.com/ArduPilot/ardupilot/commit/6dce39cbe194685e1cc021ec61a498b52c52989c#diff-a167e86094efaeeb05200b89c73ba106L702

rmackay9, 2020-07-26:
@faerietree,

In case it helps we have a wiki page here which describes how to fly without an RC. https://ardupilot.org/copter/docs/common-gcs-only-operation.html