ArduPilot/ardupilot #1749: Expand Ch7/8 Aux-Function Channels

R-Lefebvre, 2015-01-07:
There are many possible functions which can be assigned to Ch7/8 aux-functions, but only 2 channels possible to control them.  There are also plenty of low-cost RCTx units available with more than 8 channels.  We should add at least 2 more possible channels for Aux Functions.

Ideally, Aux-Function control should move into a common-vehicle library.  But at the least for 3.3, I'd like the existing Ch7/8 function expanded to 9 and 10.

multigcs, 2015-01-19:
my patch for v3.2
http://www.multixmedia.org/test/ardupilot-ch9-ch12.diff

R-Lefebvre, 2015-01-19:
Nice.

robustini, 2015-01-25:
Thanks multigcs! When in the master? ;-)

rmackay9, 2015-03-17:
This is in master now.  We may have accidentally not given proper credit to multigcs.  I apologise profusely for that.  Rob's changes are a superset of yours but I could have set the name on one of the patches to be multigcs.  Hmm, really sorry about.  I'll try and make it up to you some how.

closed 2015-03-17 by rmackay9