ArduPilot/ardupilot #1755: TradHeli: Circular swash-plate limits

jolyboy, 2015-01-09:
As discussed, request to add circular swash-plate limits for TradHeli -- @R-Lefebvre 

Current PR is located here  #1754
***** I am totally unable to compile or upload (currently at sea) so I would really appreciate your help verifying the integrity of this code.

jolyboy, 2015-01-10:
As a side note, this code did compile and bench test OK as part of the previous PR. 

This PR uses a freshly rebased master and cherry picked commits, so needs verification that nothing has gone pear shaped in the process.

R-Lefebvre, 2015-10-13:
Rebased and PR'd here:

https://github.com/diydrones/ardupilot/pull/2970

R-Lefebvre, 2015-12-15:
Closing this, fix is merged into Master and AC3.3.

closed 2015-12-15 by R-Lefebvre