ArduPilot/ardupilot #1758: Copter: YAW "angular speed" controller

robustini, 2015-01-09:
It would be very useful to control the YAW with a parameter that defines its angular velocity.
When shooting in flight the yaw movements are always too sharp, using this parameter could have a  linear motion (and smooth), which joined the "gimbal follow mode" would offer the possibility of super smooth movie.
For example: "YAW_AS", when "0" use the rate controller, when setting a value will use that speed in degrees/second (proportional, for example if you give 50% of the command the speed will be half).
It would not be useful in all the manual flight mode, like Stabilize and Acro.

R-Lefebvre, 2015-01-09:
Marco, if I understand what you are asking for, we already have this.  It's a bit counter-intuitive, but Acro_Yaw_P controls the angular rate in Acro. Default is 4.5, which gives 4.5*45.00=202.5 deg/sec.  Angular rate in other modes is controlled by ATC_Rate_Y_Max, defaults to 90.00 deg/sec. Angular acceleration of yaw is controlled with the ATC_ACCEL_YAW and you'll want ATC_Rate_FF_ENAB set to 1.

You can make the yaw very smooth with these parameters.  However, you didn't like their action on Roll/Pitch which is why it all defaults to Off. (ATC_ACCEL_YAW is Zero.)  I would suggest trying ATC_ACCEL_YAW at maybe 36000-54000.

robustini, 2015-01-09:
Thanks Robert, but ATC (which I personally use) would remain enabled in all flight modes, and would be quite inconvenient.
Many users don't use ATC, in fact it's disabled by default.

R-Lefebvre, 2015-01-09:
So are you looking for the ATC to be enabled on a Ch7/8 switch or something?

And do I read correctly that you are now using ATC?

robustini, 2015-01-10:
Yes i use ATC but it's okay to make a very thorough flew, not to shoot video.

R-Lefebvre, 2015-01-10:
Ok, then I don't understand what it is you're looking for.

lthall, 2015-01-10:
Sorry Marco, I have no idea what you are trying to describe.

Currently the yaw stick controls the yaw rate or speed. Can you try again to describe what you want.

R-Lefebvre, 2015-01-10:
Yes, we've had yaw rate control for a long time.  And now we have yaw acceleration control. I'm not sure what would be left wanting.

Maybe he's still getting a Jerk, which could partially be attributed to the non-linear torque-response you talked about.

lthall, 2015-01-10:
No, The RC_Feel limits the jerk as well. I think Marco is asking for something else.

robustini, 2015-01-10:
I asking a "linear yaw rotation speed" related to x°/sec.
For example with this parameter set to "10" one complete drone rotation in 36 seconds.

R-Lefebvre, 2015-01-10:
Oh, you mean you want like a Ch7/8 switch that makes it do this?

How about dual rates on a switch, to drop the yaw speed to the desired speed, and then just hold the stick over?

robustini, 2015-01-10:
Yes, able to yawing the drone linearly, as for the Circle declares an angular velocity, same thing but applied to the drone yaw rotation.
Today I did other tests with ATC but I did not get the expected results, the yaw unfortunately tends to swing a bit during rotation.

robustini, 2015-01-10:
Here an example, in my video.
In this case the drone rotates at more than 360°/ sec, I would like for example with all yaw command rotates perfectly at 10°/ sec, precise, and which maintains this rotation speed.

https://www.youtube.com/watch?feature=player_detailpage&v=i5q1iM6_37U#t=168

R-Lefebvre, 2015-01-10:
In that video, you are holding it hard over.  Set a switch giving you 10% Dual Rates on yaw, and then do the same, it will be much slower.

You can also use Circle mode with the Radius set to zero.

robustini, 2015-01-11:
No, I tried to reduce a lot using the end point with dual rate but the movement is not absolutely fluid, for this reason the software should be based on actual speed regulated by degrees/second.
With the Circle i can't move the drone where I want.

lthall, 2015-01-11:
The reason we are confused Marco is that is exactly what is happening at the moment.

I have been working on the yaw controller and have made some BIG improvements (along the same lines as John's work). So this may fix your issue. I am working on setting up Autotune for Yaw as well before I make it available.

robustini, 2015-01-11:
Really nice leonard, keep us informed, thanks!

robustini, 2015-01-12:
I brought this issue to the evidence in #1635, I hope we can fix.

robustini, 2015-06-27:
Yaw controller on V3.3 RC6 work fine...

closed 2015-06-27 by robustini