ArduPilot/ardupilot #1759: Feature request: Two stage Battery Failsafe

ronhogwarts, 2015-01-10:
It would be useful for pilots who fly long distances to have a two stage battery failsafe, with separate voltage/mAh settings for each. For example if I was flying 1km away and the first battery failsafe kicked in then if it was set to a fairly low voltage, say 10V (3S), then it may not be able to complete the RTL (not sure exactly what would happen in the case, potentially fall out of the sky).

With a two stage battery failsafe, if the multirotor was returning home and hit the second failsafe then it could land for example.

R-Lefebvre, 2015-01-10:
Yep, I've mentioned this need in the past.  First stage is a "Warn", second stage means "Critical".  I'll assign it to myself, but if any other devs want to take it on, be my guest.

frednfernandes, 2015-01-10:
Or, the second stage means "Caution", as is a term used on full size planes.

iskess, 2015-01-11:
I believe MP has the ability to set custom warnings on any parameter using the CTRL+F window. I'm not near a computer now to try.

WickedShell, 2018-03-10:
Presented in #7213 (tagging so I remember to come back and close this)

rmackay9, 2018-03-27:
Great stuff @WickedShell!

Zekeortiz, 2018-06-05:
Is it a possibility to have more than just RTL or Land , but both like a secondary battery failsafe? For example a quadcopter could have an initial battery failsafe set to rtl at 14volts, and a secondary battery failsafe set to land at 13.3 volts. That way if the drone can't make it to the home point with the remaining battery that is left then it would simply land and not fall from the sky half way to the homepoint and prevent damage.