ArduPilot/ardupilot #1762: Feature request: information on the screen for the reason of failsafe

gtiramani, 2015-01-11:
Mission planner gives the message of failsafe, but not specified the anomaly that has activated: it would be very useful to have immediately detailed information with the parameter and the eventual value

rmackay9, 2015-01-12:
Thanks for the report.

We send messages to the GCS which clarify if the the failsafe is not simply a throttle failsafe but perhaps the GCS is not displaying all the messages.  I.e. perhaps the big Failsafe message is overwriting the secondary descriptions.  In that case this might be best handled in the GCSs instead of in arducopter.

This is what is sent from arducopter: 
Throttle failsafe: "Failsafe" will appear
Battery failsafe: "Failsafe" and "Low Battery!" should appear
GPS failsafe: "Failsafe" and "Lost GPS!" should appear
EKF/DCM failsafe: "Failsafe" and "EKF Variance" will appear

We have another to-do item somewhere on the list to replace the messages with numbers.  Perhaps that would make it easier for the GCSs to print the highest priority message.

Naterater, 2019-01-05:
I think this has been implemented, albeit it only sends failsafe type and reason as numbers, not explanations. The ground stations should probably be decoding the reasons. See https://github.com/ArduPilot/MissionPlanner/issues/1855

closed 2019-01-05 by Naterater