ArduPilot/ardupilot #1768: Check for compass orientation errors on startup

proficnc, 2015-01-12:
It is possible to install a magnetometer on a pixhawk facing the wrong direction.  If you then complete all the checks, it will pass prearm, and will fly ok in Stabilise.  Unfortunately when you switch to an Auto mode, the system will not have a reliable Yaw reference, that could lead to an unpredictable flight condition.
The EKF can notice this situation, but there is no action taken that would stop the copter going into an Auto mode.

Two suggestions:- 
1 that we modify the pre arm to detect compass mismatch.
2 that we add a step to the compass calibration on the GCS's to verify the orientation of the compass, and other sensors before allowing the calibration to pass.. (planner issues, if anyone likes this suggestion, I will raise an issue with the GCS's)

https://drive.google.com/folderview?id=0BzZ2pkkTrPezczB0dk5LLUItMXc&usp=sharing 

the file is compass_error_clean.log

amilcarlucas, 2018-10-19:
the new in-flight mag calibration PR allows you to get this done.

IamPete1, 2021-08-08:
We now have auto orientation of compasses and the GSF for error detection. 

https://www.youtube.com/watch?v=3xW9hj-lxNU

closed 2021-08-08 by IamPete1