ArduPilot/ardupilot #1772: How to compile the Arduino code ?

adriensupizet, 2015-01-13:
Hello, 
We're a group of 6 second year students studying engineering in France.
We're working on an annual project which consists in building our own quadrotor using ardupilot.
We have downloaded all the files needed and we're having a hard time trying to understand how to compile it with the Arduino IDE.
We have followed all the steps indicated on the arduino.cc website but we still face compilation issues.

We need your help to make us understand where we're doing it wrong.
 
   Thank you for your time, we hope that you will be able to help us.

santosh-mahadev, 2015-01-13:
I am also searching a tutorial from where we can learn how to start development on ArduPilot

shpala, 2015-01-13:
This is a good starting point for programmers who want to dive deep into the code:
http://dev.ardupilot.com/wiki/apmcopter-programming-libraries/

squilter, 2015-01-13:
dev.ardupilot.com has lots of good resources.  I wouldn't recommend using the arduino IDE any more.  In my opinion, the easiest place to start would be [building for pixhawk using make in linux](dev.ardupilot.com/wiki/building-the-code/wiki/building-px4-for-linux-with-make/)

closed 2015-01-13 by squilter