ArduPilot/ardupilot #1775: Compiler arm-none-eabi-g++ (4.8.3-11ubuntu1+11) gives stack frame size errors - why?

gmorph, 2015-01-14:
/home/grant/3dr/ardupilot/libraries/AP_NavEKF/AP_NavEKF.cpp: In member function 'void NavEKF::CovariancePrediction()':
/home/grant/3dr/ardupilot/libraries/AP_NavEKF/AP_NavEKF.cpp:1816:1: error: the frame size of 2264 bytes is larger than 1024 bytes [-Werror=frame-larger-than=]
 }
 ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
/home/grant/3dr/PX4Firmware/makefiles/module.mk:209: recipe for target 'libraries/AP_NavEKF/AP_NavEKF.cpp.o' failed
make[3]: *** [libraries/AP_NavEKF/AP_NavEKF.cpp.o] Error 1
make[3]: Leaving directory '/home/grant/3dr/PX4Firmware/Build/px4fmu-v2_APM.build/home/grant/3dr/ardupilot'
/home/grant/3dr/PX4Firmware/makefiles/firmware.mk:236: recipe for target '/home/grant/3dr/PX4Firmware/Build/px4fmu-v2_APM.build//home/grant/3dr/ardupilot/module.pre.o' failed
make[2]: *** [/home/grant/3dr/PX4Firmware/Build/px4fmu-v2_APM.build//home/grant/3dr/ardupilot/module.pre.o] Error 2
make[2]: Leaving directory '/home/grant/3dr/PX4Firmware/Build/px4fmu-v2_APM.build'
/home/grant/3dr/PX4Firmware/Makefile:128: recipe for target '/home/grant/3dr/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4' failed
make[1]: *** [/home/grant/3dr/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4] Error 2
make[1]: Leaving directory '/home/grant/3dr/ardupilot'
../mk/px4_targets.mk:76: recipe for target 'px4-v2' failed
make: *** [px4-v2] Error 2

bugobliterator, 2015-02-19:
The cap of 1024B stack-space is because of the following line in makefile for building PX4Firmware, changing this value should do the trick:
https://github.com/diydrones/PX4Firmware/blob/master/makefiles/toolchain_gnu-arm-eabi.mk#L141

gmorph, 2016-02-11:
We have upgraded to 4.9.3 so this is no longer relevant.

closed 2016-02-11 by gmorph