ArduPilot/ardupilot #1782: when I build ArduPlane with "sudo make px4-v2",an error occur:" cp: cannot stat '/../mk/PX4/config_px4fmu-v2_APM.mk': No such file or directory"

zhxzsh, 2015-01-16:
When I build ArduPlane with "sudo make px4-v2",px4io-v2_default.px4 is generated successfull,but when build px4fmu-v2,an error occur:" cp: cannot stat '/../mk/PX4/config_px4fmu-v2_APM.mk': No such file or directory",why?My OS is ubuntu 14.04 lts.

gmorph, 2015-01-16:
Hi.  You shouldn't be running make as root so don't sudo.  Second you need to be in the ArduPlane directory and run "make configure" first and then "make px4-v2".  Finally there are some per-requisites you need need to have installed.  This link has a bunch of info.
http://dev.ardupilot.com/wiki/building-the-code/
Thanks, Grant.

closed 2015-01-16 by gmorph

zhxzsh, 2015-01-16:
"You shouldn't be running make as root so don't sudo" ,when I running make as normal user,error:"/bin/sh: 1: cannot create /home/zhx/ardupilot/Build.ArduPlane/make.flags: Permission denied"

zhxzsh, 2015-01-16:
Thanks gmorph!All successfull!The source reason is file limitation.change all file attribution to default user,all successfully OK!