ArduPilot/ardupilot #1783: New Chalenge for Pixhawk Programmers  "Copter-Jet"  Mix between Copter and Plane

lorbass, 2015-01-16:
Copters need hight Power. Planes a lot of space for start and landing.
Imagin the following:
A Delta Plane, 2 Props (or 4) at the wings (in front) 
Vertical start, hoovering like a copter, stabilized one axis with the two props, the other axis with rudders.
More pitch, moving like a copter, more, and more pitch the delta goes to the horizontal and flys like a plane wiht speed and low power.
The same back the pitch stick to Zero. Return to hoover, and land verticaly.
Not far away from Arducopter or Arduplane.
This would be a real innovation, not only up and down of the arms like ....
Usefull for long range FPV or Parcel transportation.
What ist the meaning of professionals like Marco or Randy?

lorbass, 2015-01-16:
I mean this
http://www.flightglobal.com/news/articles/google-unveils-uav-parcel-delivery-concept-403160/

Craig3DRobotics, 2015-01-16:
Please see https://github.com/Tiltrotor

lorbass, 2015-01-17:
Thanks for the reply but I don't mean Tiltrotor or Tiltwing.

R-Lefebvre, 2015-01-17:
Yes, but it's pretty much all working in the same direction.

JSGordon, 2015-01-28:
I think this is another up and coming direction in platforms.  It can combine the range and efficiency of fixed wing with the versatility of deployment of multicopters VTOL.  Google and a few others have fooled with this.  Most of the current implementations seem to use two flight controllers and management unit to handle the transition.    The big issue is to manage a transition from VTOL multicopter type control to fixed wing and back.  Doing it without tilt rotors is a mechanically simpler solution compared to tilt rotor and probably more reliable. Having a single flight controller handling all flight modes would be nice.  
An approach where only VTOL and position hold are in multicopter mode and all other navigation  is done as in Arduplane.  I think this type aircraft would spend little time in multicopter mode typically and any multicopter mode use would not typically involve much forward flight as the wing structure would make this extremely inefficient.

Would be willing to make a test platform if any development begins on this

lorbass, 2015-02-01:
You are right.
I'm shure, it is not easy. But as I judge the SW Freaks of the Ardu FM they arn't looking after simple things. Especially if large housis had no success.
Of course we should not want to hoover to the moon, but I' m convinced this will be the next generation of FC. Somewhere in West oder East will bring it. Since FPV is now possible up to 1.7 km with FullHD Quality we need more Flighttime witch is possible with fixed wing. In region of interrest it is mostly not possile to start or land with planes.
Here the first project with succes but with too much of "Hardware"
http://www.birdseyeview.aero/

lorbass, 2015-02-03:
Watch this: https://www.youtube.com/watch?feature=player_detailpage&v=IK_-yTrwNtU
and read the 517 comments and you understand my "inspire".
Already realized.

WickedShell, 2017-11-04:
Quadplane supports this.

closed 2017-11-04 by WickedShell