ArduPilot/ardupilot #1799: Plane/Copter Support for temperature sensor

mirko91, 2015-01-20:
Is it possible to add a support for an analog temperature sensor like an LM35 just to monitor the ESC or air temp and log the value or even send it to the Planner? Cheers

rmackay9, 2015-01-20:
we send a temperature from the barometer.  it can be seen as raw_temp or press_temp in the mission planner's flight data > status tab.   maybe you can check those and if they're sufficient we can close this issue.

mirko91, 2015-01-20:
Hi Randy, I know that but the temperature readed by the barometer it is higher than the effective because of the board heat. I mean it is a nice addition if it isn't too much an hassle.

CodeChief, 2015-01-20:
I'll second that please. It'll be something like a shrink-wrapped thermistor, an A2D converter and a circuit to plug it into the I2C bus. I suppose one of those tiny ATMEGA dev boards would do, like people use when converting telemetry. Oh yes...

On that subject maybe that is the solution. A new generic sensor is new telemetry of some kind. Why re-invent the wheel then. All you have to do is be able to read the receiver's telemetry system sensor on the Pixhawk via it's SBUS, e.g. FrSky 8XR SBUS connected with this... http://www.hobbyking.com/hobbyking/store/__19976__FrSky_TEMS_01_Telemetry_Temperature_Sensor.html

Of course in the long run popular bits and pieces of hardware should be mass produced by 3DR and free to clone so generally available (e.g. HK Pilot). A temperature sensor makes sense, very basic and universal. Think about ESC overheat/burnout protection, auto-safe modes and things like that in the future, nice!

mirko91, 2015-01-20:
Glad you find it useful. I think it is a nice plus and a cleverer failsafe. I think a cheap atmega 328p could take care of the A/D conversion and read up to 8 analog sensors and than send it to the i2c bus for a one plug connection.

multigcs, 2015-01-25:
i find it useful to send the raw analog-values from ADC's to ground and a simple config on gcs (offset/scale/unit).
so you can customized and display all possible sensor data you want

-- olli

ps: sorry for my english :)

mirko91, 2015-01-25:
That is a good starting point but I think it could be more powerful, but for now it'll be plenty

samucs, 2018-11-14:
+1 Please, could you add some priority to implement this? We wish to monitor the temperature of medicines during flight. Also, some vaccines will turn into waste because of temperature fluctuations. We are also considering a donation to the dev team to get this done. All we need is to read temperature and humidity values from sensors like DHT11 and stream the values over Telemetry to the GCS.

vierfuffzig, 2018-11-14:
@samucs  additional temperature monitoring is well possible based on current code using an external barometric sensor: http://ardupilot.org/copter/docs/common-baro-external.html
it won‘t provide monitoring humidity though.
https://github.com/ArduPilot/ardupilot/issues/9567

auturgy, 2018-11-17:
> i did some testing with TMP36 type thermistors, but found them to be far too much prone to RF induction to be used in our kind of application. as soon as anything RF is used within 2 meters perimeter of the sensor / wiring, even if properly shielded, there‘s just no reliable values anymore.
> imho a typical baro is just as affordable, well tested and reliable, plus easy to add based on existing prerequisites for auxilliary temperature telemetry and OSD display.

One other option is Ollie’s uavcan baro node, which will convert a bmp280 or ms5611 to a can device - overcoming the issues with long i2c connections, or i2c bus conflicts.
http://www.olliw.eu/2017/uavcan-for-hobbyists/#chaptergspmag
You can buy one from jdrones now. 

Regards,

James

grcasanova, 2018-11-18:
I'm testing a NTC 3950 and a ferrite ring and the collected data is quite reliable, consistent with an external temperature sensor whilst not very precise (due to the sensor's specs).

deverejohn, 2018-11-29:
+1 Would love to be able to easily monitor the embedded lm35 in my LiPo’s.

Naterater, 2019-01-18:
Does this conversation belong here: https://github.com/ArduPilot/ardupilot/issues/877?

IamPete1, 2021-08-14:
We now have analog inputs in scripting and the ability to send named value floats, these can be disaplayed with the correct name by both Mission planner and MAVProxy.

Analog input example here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/analog_input_and_GPIO.lua

Sensor reporting back using named floats here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/SN-GCJA5-particle-sensor.lua

closed 2021-08-14 by IamPete1