ArduPilot/ardupilot #1808: Copter: get back mount2

nhadrian, 2015-01-22:
Hi,

I'd like to ask to get mount2 back into APM:Copter.
The reason is the following, can be really useful for a combined purpose FPV copter.

I'm using mount2 for frame position input for BLS gimbal controller. Why? Because 8-bit bls controllers doesn't have a secondary frame sensor. So when using follow-mode in roll and pitch axes, there is a significant drift in position during a complete flight. But when I connect the roll/pitch stabilized APM outputs to the controllers frame inputs, the frame following works for perfect for a long time. So technically I get a smooth camera frame following which is really useful in FPV.

But why not connect it to mount1? Because I use the camera not only for FPV live view but taking HD videos too, sometimes in a single flight. But when roll and pitch is stabilized, I can't let camera ignore frame movements and control position accurate enough.
So I connected mount1 APM pitch output without stabilization to the gimbal controller's RC-pitch input, this means I can control the pitch, or let APM control it for a POI capture in auto mode.

I use a switch to control the gimbal controller mode, that means I can turn follow on/off inflight. 

Thanks in advance.

Regards
Adrian

ps.: I tested this method with APM and Pixhawk too, works great with 3.2 and mount2 enabled.

R-Lefebvre, 2015-01-22:
Wow, very interesting use case we hadn't considered.

nhadrian, 2015-01-22:
Here is a small sample video of the combined modes:
https://www.youtube.com/watch?v=tqoR1SrphTs

The best would be to have Mount2 enabled by default, but it is not as essential, I can build the code for myself.

nhadrian, 2015-01-22:
Here is a modified connection diagram according to my solution
[image]

R-Lefebvre, 2015-01-22:
Yep, I totally get what you're saying.

Did Randy just turn off the Mount2 #define or remove it altogether?

R-Lefebvre, 2015-01-22:
Technically, what you really want though, is a simple Attitude-to-PWM output.  It could be a very simple subset of the full Mount object.

nhadrian, 2015-01-22:
Mount2 #define was DISABLED by default, but he removed Mount2 completelly some commits ago.

nhadrian, 2015-01-22:
Also, I found that when FC is connected and configured properly, really makes gimbal stabilization even better and more accurate especially during high angles or fast movements.

nhadrian, 2015-01-22:
Removed here: https://github.com/diydrones/ardupilot/commit/b6bcf841e1e2f41ec89858defd21b62bc675e5c4

rmackay9, 2015-01-23:
This is great information.  I guess you're using the AlexMos gimbal?

I removed Mount2 but only because it's being replaced by an improved frontend-backend split which allows completely different mount libraries to be used based upon a MOUNT_TYPE parameter.  So we will have a servo type, a MAVLink-enabled-gimbal type and an AlexMos type.  There may be a SToRm gimbal type eventually as well.

It sounds like a more strategic solution would be to add two more outputs from the servo-lib to send the frame angle (via PWM) to the gimbal.  Alternatively if this is an AlexMos gimbal then we can send that through the serial interface.  It's really great that you've figured this out, let's take that and make it easy for other people to setup.

rmackay9, 2015-01-23:
One extra thing is we could add a ch7/ch8 option to control the gimbal mode.

nhadrian, 2015-01-23:
Hi,

yes, I'm using an Alexmos Gimbal.
I think PWM is better than serial, since with a complete enough setup there won't be any free serial port (ie. using telemetry module, OSD, two GPS modules…). PWM is simple and can be configured easy in Basecam GUI, the range is detected automatically by moving copter around for a few times.

A simplified solution could be great, I came up with some ideas:
For example make selectable "gimbal_frame_roll" and "gimbal_frame_pitch" items under RCxx funcitons, the only parameters should be configurable is the PWM mid value, the PWM range and the direction. So fo example: 
- GIMBAL_FUNCTIONS_ENABLE - enable or disable the complete gimbal control - default 1
- GIMBAL_FRAME_ROLL_MID - mid point in us - default 1500
- GIMBAL_FRAME_ROLL_SENS - degrees to PWM ratio - default 4,5 that means 405 us PWM for 90 degrees of bank
- GIMBAL_FRAME_ROLL_REVERSE - the directions - default 1 but can be -1
- GIMBAL_FRAME_PITCH_MID - mid point in us - default 1500
- GIMBAL_FRAME_PITCH_SENS - degrees to PWM ratio - default 4,5 that means 405 us PWM for 90 degrees of bank
- GIMBAL_FRAME_PITCH_REVERSE - the directions - default 1 but can be -1
Automated gimbal mode control is also a great possibility, that means copter could switch from follow mode to fixed mode during a mission. The only thing needed is a free output channel.

For example, make a selectable item in RCxx functions for "gimbal_control_channel"
Parameters:
- GIMBAL_MODE1_VALUE - output value for mode1 in us - default 1100
- GIMBAL_MODE2_VALUE - output value for mode2 in us - default 1900
- GIMBAL_MODE_DEFAULT - default mode to be used in manual flights - 1 or 2
- GIMBAL_CONTROL_INPUT - input RC channel for manual override of gimbal modes - from 7 to 10

And make a new mission item:
- DO_GIMBAL_MODE - switch gimbal mode to 1 or 2

Regards
Adrian

ps.: I'm using this on my plane too, so once a library is modified, should also be used in plane!

R-Lefebvre, 2015-01-23:
That's a great idea Adrian.

Randy, would this fit into what you're doing with front end/back end split?

These RC_Channel_Aux functions could be added to the existing Mount library pretty easily, so that they would be available without needing a full second Mount object.  If you want to assign this one to me I can do it.  Fairly straight forward.

rmackay9, 2015-01-23:
In general I think if this is for AlexMos we should first try and implement it with the upcoming AlexMos mount library.  We should be able to do it that way and adding more complexity to the existing servo lib doesn't seem like a great idea to me and why have two solutions to the same problem?  If we have an AlexMos gimbal library, let's make it as good as possible so people us it.

The SerialManager that goes with it will allow more flexibility in assigning serial ports which should help alleviate some of the traffic.

R-Lefebvre, 2015-01-23:
So you want to do it all over serial instead of PWM?

nhadrian, 2015-01-23:
I'd vote on PWM since it is simple and robust method, and you don't have to worry of firmware or serial version mismatch between different controller firmware types. Also other type of controllers (like Martinez) could be used.

R-Lefebvre, 2015-01-23:
I agree.

One consideration is that PWM is known to work today, with a very high "just works" factor.  Serial comms are slightly in the future, and we don't really know how well it will work out, mostly due to our lack of control over the 3rd party gimbal software.

nhadrian, 2015-01-23:
OFF: could anybody please send me a compiled .hex ArduPlane 3.2 firmware with MOUNT2 enabled to the nhadrian@gmail.com? With this new style of GitHub client for windows I can't figure out how to revert to ArduPlane 3.2 tag… :( I only see the branches but not the tags...

nhadrian, 2015-02-05:
PLEASE get back mount2 define possibility into ArduPlane too.
I can't use my current setup without mount2, since mount2 controls the frame-follow mode (as described above). Many thanks in advance!!!!!!

rmackay9, 2015-02-06:
@nhadrian, two mounts are still supported.  The issue may just how it's implemented.  There's no longer a camera_mount2 object but instead the single camera_mount object has multiple instances within it.
To define that you need two mounts go into the AP_Mount/AP_Mount.h file and change the AP_MOUNT_MAX_INSTANCES to 2.
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Mount/AP_Mount.h#L34

nhadrian, 2015-02-06:
OK Randy, it works!!! :)
I suggest to update the MOUNT_TYPE_x and DEFLT_MODE_x descriptions in MissionPlanner, quite difficult for an end-user to find the description in the AP_Mount.cpp. Thanks in advance!

Regards Adrian

rmackay9, 2015-02-06:
@nhadrian,
I've already updated the descriptions for the mount type and DEFLT_MODE (shown below).  Would some other wording be more clear?
_TYPE:
// Description: Mount Type (None, Servo or MAVLink)
// Values: 0:None, 1:Servo, 2:MAVLink, 3:Alexmos Serial
_DEFLT_MODE:
// Description: Mount default operating mode on startup and after control is returned from autopilot
// Values: 0:Retracted,1:Neutral,2:MavLink Targeting,3:RC Targeting,4:GPS Point

nhadrian, 2015-02-06:
No that is OK. But the cell was empty near the parameter in the missionplanner parameters tab. I have the latest MissionPlanner beta available as an update.

rmackay9, 2015-02-06:
@nhadrian, ok great.  Yes, it's a common issue that the MP parameters are not up to date.  out of my control.

R-Lefebvre, 2015-12-21:
It seems like this issue is closed. Please reopen and comment on current status and what needs to be done if it's not.

closed 2015-12-21 by R-Lefebvre