ArduPilot/ardupilot #1815: Plane/Copter: two separate waypoint mission, not only one

robustini, 2015-01-23:
It would be very interesting to have the chance to perform in flight two separate missions, such as "AUTO1" and "AUTO2".
This would open up many interesting scenarios, for example with Plane a mission (AUTO2) that leads the aircraft on track with automatic landing.
Interesting is also the possibility to perform a "GO TO AUTO2" at the end of "AUTO1" or vice versa.

robustini, 2015-01-23:
Simpler alternative: have the possibility via parameter to perform only a part of the mission by the event number, for example:

AUTO -> complete the mission
AUTO2 -> execute the mission from a "particular wp event" (previously marked as "separator")

dellarb, 2015-02-02:
Hi Rob - I've implemented this using jump commands with existing mission architecture. 

Mission 1 was a series of waypoints with a jump back to start of mission 1 at the end and the next mission began after that. To start the second mission I would need to set the waypoint # to be one at the start of that mission. Similarly you could use the jumps to move between them based on how many iterations had occurred etc.

Is that what you are after?

robustini, 2015-02-02:
@dellarb: yes, but there must be a way to run them separately via tx.

WickedShell, 2019-01-05:
Yes.

auturgy, 2019-01-05:
This is actually a great use case for scripting @wickedshell

robustini, 2020-06-06:
> Hi Rob - I've implemented this using jump commands with existing mission architecture.
>
> Mission 1 was a series of waypoints with a jump back to start of mission 1 at the end and the next mission [...]

In many cases with Plane it would be interesting to have the possibility of carrying out the second mission set, for example in the event of a failsafe, in order to perform an automated landing with a preset route.
Then Mission 1 for the desired route, and Mission 2 in case of failsafe or always from radio switch.
With the Copter it might be irrelevant, but very useful with the Plane.

robustini, 2020-06-06:
In the event of a tx failsafe the Plane returns to the Home but landing it automatically without a pre-planned route is somewhat difficult in many cases.
Copter and VTOL would have no problem, but the Plane would.

WickedShell, 2020-06-06:
Marco you know about DO_LAND_START right? The plane can jump to the nearest
one of those too do a landing (a valid failsafe option for most things)
then it does all the waypoints after it in order. You can even plan
multiple do land starts and the plane just picks the one closest to it,
which gives you multiple predefined return paths.

robustini, 2020-06-06:
Of course, but it would be more useful to have a second mission also for other reasons.

closed 2020-06-06 by robustini

magicrub, 2020-06-06:
This is actually a gcs issue, as multiple missions can easily be stored as a linear list to jump around in. However, there are things the autopilot could do to make this easier like adding markers, similar to DO_LAND_START. We could make a mission_start label that does nothing for the autopilot but a gcs could use it to visualize multiple missions. Piccolo does this very well. I would be very happy to see such a feature in ArduPilot. I often have to fly with secondary and other pre-planned scenarios that I need to execute quickly without having to upload to switch between. Would be nice to say "set mission 3" and it does it

peterbarker, 2020-06-07:
On Sat, 6 Jun 2020, Tom Pittenger wrote:

> This is actually a gcs issue, as multiple missions can easily be stored as a linear list to jump around in. However, there are things the autopilot could
> do to make this easier like adding markers, [...]

There are already mavlink commands to do this, which is cool.  They look 
pretty reasonable, too.

       <entry value="600" name="MAV_CMD_JUMP_TAG" hasLocation="false" 
isDestination="false">
         <description>Tagged jump target. Can be jumped to with 
MAV_CMD_DO_JUMP_TAG.</description>
         <param index="1" label="Tag" minValue="0" 
increment="1">Tag.</param>
       </entry>
       <entry value="601" name="MAV_CMD_DO_JUMP_TAG" hasLocation="false" 
isDestination="false">
         <description>Jump to the matching tag in the mission list. Repeat 
this action for the specified number of times. A mission should contain a 
single matching tag for each jump. If this is not the case then a jump to 
a missing tag should complete the mission, and a jump where there are 
multiple matching tags should always select the one with the lowest 
mission sequence number.</description>
         <param index="1" label="Tag" minValue="0" increment="1">Target tag 
to jump to.</param>
         <param index="2" label="Repeat" minValue="0" increment="1">Repeat 
count.</param>
       </entry>

Not sure I like that "multiple matching tags" resolution.  Haven't given 
it much thought.

Perhaps a good-first-issue?

robustini, 2020-06-07:
> This is actually a gcs issue, as multiple missions can easily be stored as a linear list to jump around in. However, there are things the autopilot could do to make this easier like adding markers, [...]

Absolutely yes, it would be very useful in many scenarios.
And if you could associate a FAILSAFE event with a specific mission it would be fantastic.
However, in this case it would be enough to associate this event with a particular tag in the list of stored IDs, for example in the case of RX FAILSAFE browse the stored EVENT_IDs and when find the ID "RECOVERY" execute all the IDs stored after that, easy.