ArduPilot/ardupilot #1819: Plane new land flight mode

mirko91, 2015-01-26:
I'm thinking about a new Land flight mode that will work the following way:
-one parameter to set if the landing spot will be the same as the home location or a new one
-if it is the same as the home the plane will do 4 loiter turns at 40m AGL with the home on the centre like RTL and calculate the wind direction and fly 200m away from the spot and land against the wind
-else if the land location is not the home point it'll be the point where the mode will be triggered, than the land procedure will be the same as above

I think that with this method the landing will be more accurate and fully autonomous

tridge, 2015-02-05:
Hi Mirko,
I think we need to first document and complete the DO_LAND_START support that was added recently. That isn't the same as what you propose, but it does offer a flexible way of setting up auto lands.
I'll add this as something to be reviewed for 3.2.2
Cheers, Tridge

mirko91, 2015-02-05:
Thanks Tridge, 
Really appreciate what the APM team is doing! Looking forward to see what DO_LAND_START is capable of. Cheers!

Naterater, 2018-12-01:
Landing is a part of AUTO mode and has been fully implemented and mostly documented.

closed 2018-12-01 by Naterater