ArduPilot/ardupilot #1820: Unable to connect to Mission Planner with AC(V3.3, 3.2.1 ,3.2)

Tuck3rz, 2015-01-26:
Good evening, I have an APM2.6 and I have downloaded the Arducopter codes from github and uploaded it to APM via USB (with nothing connected to it) using Arduino IDE. 

These codes are uploaded successfully (with CLI enable in APM_Config.h). However I'm not able to connect them through Mission Planner and all i got was "No heartbeat packets received". I checked it through the terminal and no codes were running in the CLI as well.

I have tested it with v3.1.5 and it managed to connect to Mission Planner though.
Does anyone have any idea why the later version does not work for me?

rmackay9, 2015-01-27:
This looks like more of a support question in which case it's probably best to raise in the APM Forum.
http://ardupilot.com/forum/viewforum.php?f=100

In any case, I'm afraid master won't fit on an APM2.x.  We're not supporting the AVR CPU based boards beyond AC3.2.1.

closed 2015-01-27 by rmackay9

Tuck3rz, 2015-01-27:
Hi there, I have tried using the branch files of 3.2 & 3.2.1 too but it does not connect as well