ArduPilot/ardupilot #1822: Create PWM input driver

gmorph, 2015-01-27:
Create a PWM input driver as an alternative for talking to PulseLight LIDAR instead of using I2C.

rmackay9, 2015-01-27:
I'm not a big fan of this idea.  I think the LIDAR-Lite has problems with it's I2C interface and those should be resolved by the LIDAR-Lite guys.  Because we talk with them regularly and have a generally good relationship I think we should push them to resolve the issues (and I think they will resolve them).

Using the PWM interface is a step backwards because we would not be able to determine whether the lidar-lite is healthy or not  (and other things like what version it is, etc).  It also adds complexity by allowing two different methods to use the LIDAR-Lite which is confusing for users.

Worst case I'd say if we don't believe in the I2C inteface for the LIDAR-Lite and we don't think it'll ever be fixed we should remove the I2C library and replace it with the PWM version.

tridge, 2015-01-27:
I'd like a PWM driver for a few reasons:
* I'd like to be able to add other PWM based sensors, and this is a good one to get started with
* if it works we can use existing Lidar lites, without requiring everyone to buy a new one or ask PulsedLight to replace them
* we should be able to detect a healthy Lidar through PWM values
* it will be a very useful diagnostic tool for the Lidar issues, by bypassing I2C

rmackay9, 2015-01-27:
Ok, especially the 2nd point that you mention is important (people who have bought lidar lites that can't be used reliably with I2C).  Sorry to be annoying, I can just hear the support calls, "which interface should I use for Lidar lite?".  Perhaps as a documentation issue we can push people towards the I2C interface once the I2C issues are fixed.

magicrub, 2015-02-20:
@tridge , is there a status update on this?

tridge, 2015-02-20:
yes, I have proven it can work, and I have a very rough prototype driver using FMU AUX5. I hope to have something I can test fly soon.

magicrub, 2015-02-20:
I am an eager alpha tester!

tridge, 2015-02-21:
I have a prototype which I hope to test fly tomorrow

magicrub, 2015-02-22:
I don't see it in master. Can you send me your binary so I can test fly it
with you?

tridge, 2015-02-22:
You'll need these two branches:
https://github.com/tridge/Firmware/commits/pwm-input-wip
https://github.com/tridge/ardupilot/commits/pwm-input-wip
I did several flights with it today. A couple of them went very well, but others were less good with the rangefinder going unhealthy at quite low altitudes. I suspect the filtering in this code is the issue:
https://github.com/tridge/ardupilot/blob/9225e83f7723e11f031effc9a3fa5ed9a226a7ab/libraries/AP_RangeFinder/AP_RangeFinder_PWM_PX4.cpp#L102
If you want to test it, you need to connect it like this:
Lidar J1-pin1 -> 5V servo rail (say FMU AUX5 power)
Lidar J1-pin3 -> FMU AUX5 signal
Lidar J1-pin6 -> FMU AUX5 ground
you also need a resistor between the signal and ground pins. I use a 470Ohm resistor, but anything between 200 and 1k should do.
Then you need to set RNGFND_TYPE to 5.

tridge, 2015-02-23:
its in 3.3.0beta1
See instructions here:
http://plane.ardupilot.com/wiki/common-rangefinder-lidarlite/

closed 2015-05-06 by magicrub