ArduPilot/ardupilot #1823: Arducopter Binary sketch seems to be too heavy for the Arduino Mega

adriensupizet, 2015-01-27:
Hi, I'm having trouble trying to compile the Arducopter source code.
The binary sketch for the Arducopter code seems to be too heavy for an Arduino mega.

Is the lastest version too heavy ?
Or is it safe not to use lastest version, which may be heavier than a previous one ?

Is there any way we can reduce the binazy size of the sketch ?

Here's the error message we're getting at the compilation : 
     "Excluding arduino core from link
     Size of the binary sketch : 298 970 bytes ( max 258 048 bytes)
     processing.app.debug.RunnerException:
	     at processing.app.Sketch.size(Sketch.java:1658)
	     at processing.app.Sketch.build(Sketch.java:1591)
	     at processing.app.Sketch.build(Sketch.java:1567)
	     at processing.app.Editor$DefaultRunHandler.run(Editor.java:1913)
	     at java.lang.Thread.run(Thread.java:619)"

Thanks for your time, I hope you'll be able to help me

davidbuzz, 2015-01-27:
Which specific version are you using of the code?
Which specific IDE  version are you using.? 
Which specific hardware/board are you using.?

R-Lefebvre, 2015-01-27:
I'm assuming he is trying to compile Arducopter-Master for APM.  

The APM will no longer be supported in future Arducopter developments as it has reached end of life due to insufficient resources (in this case RAM).  It will be supported in bug fix releases for the 3.2.x branch only from this point.

closed 2015-01-27 by R-Lefebvre