ArduPilot/ardupilot #1827: RXRSSI bottom end calibration

JSGordon, 2015-01-28:
There is a feature now in APM which allows for a analog input from receivers such as FRsky.  Already in code for 3.2 there is the ability to set voltage input range to calibrate the 100% RSSI point.
However there is no such feature to set the low point.

The FRsky RSSI output is a PWM output that typically is integrated into a DC analog input with a resistance capacitance filter.  Since the PWM pulse train typically varies from 1000 to 2000 milliseconds the analog DC never goes to zero.  Other implementations typically allow for both top end voltage and bottom end voltages to be set based on observed values.

It would be nice if this capability existed in APM as well.  From my experience recently the usable range is compressed into what APM shows as ~60 to 100% with 60 being the point where control may be lost.

nhadrian, 2015-01-29:
Hi,this is strange for me. I'm using FRSky TFR4 FASST compatible receivers and the PWM RSSI output. I'm using R-C filter too. The RSSI range is set to 3.3V in parameters.
For me the RSSI value is between 1 and 100, so I think the RSSI PWM output varies from 0 to 2000.
Unfortunatelly I can't measure PWM since I don't have any oscilloscope.

I think a direct PWM RSSI reading would be even more useful feature, users should not build DIY R-C filter, so would be more user-friendly for all FRSky receiver users.

taedla01, 2015-01-29:
Hello everyone,

For analog RSSI input (Plane, Copter and Rover codes) I opened some time ago three pull requests that solve your problems with low RSSI value. I think it's a matter of time that they will be included by a moderator  into master branch.

You can see here:
For Plane: #1603 
For Copter: #1604 
For Rover: #1633

Regards from Spain,
Dario.

JSGordon, 2015-01-31:
Thanks Dario,
Did not see these.

nhadrian,

There are a number of different scenarios where this could be useful.
As for direct RSSI I am not sure what you mean.  A digital output from the receiver to a flight controller? I think this is unlikely as it would require receiver manufacturers to provide a digital output compatible with APM.  Currently a PWM pulse train for analog output of some type of RSSI is the most common available and people want to utilize this now.  This requires a simple filter. You can purchase cables with the appropriate LC filterto do this just like other interface cables such  as voltage dividers for voltage monitoring or build.

One scenario where APM could read RSSI without a filter is -In a OpenTX based radio such as the Taranis the RSSI can be fed as a input to a channel resulting in a standard PWM signal from 1500 to 2000 millisecond pulse width corresponding linearly to the RSSI.  So far I have not been able to get a full range PWM but with a mode to OpenTx thsi would be possible. This would be a feature that could be added and any radio capable of doing this could provide a easily usable RSSI output that could go to a flight controller via a spare channel.

Alternatively this can also be filtered and used as a analog input especially if able to calibrate top and bottom voltage levels.  This results in the RSSI as determined by the radio to be output and is consistent with the radios estimation of RSSI.  But to use both top and bottom end need to be calibrated.  

I checked the D4R today on the scope. On the D4R Frsky RSSI it outputs a pulse every 900 milliseconds that is not particularly well defined. It goes down to ~ 5 to 10 ms and up to about 890.  There is not a well defined leading edge but a significant ramp for some reason.  PWM outputs from the same receiver are very well defined square waves. When integrated this does not correspond very well to the radios RSSI figure unfortunately.  On APM I get form 5 to 15% on the bottom depending on receiver and setup.  While not perfect being able to calibrate the bottom voltage could help expand the range and correlate more accurately access link integrity.

Maybe we should put a request for the conventional PWM on a channel as a method.  Any comments?

nhadrian, 2015-01-31:
Hi 

Here you can find the detailed description of the RSSI PWM in the Frsky TRF4 manual. I hope this helps. For me it looks quite simple.

http://www.robotshop.com/media/files/pdf/manual-ac-0004-24.pdf

Regards
Adrian

JSGordon, 2015-03-06:
Interesting.  This is what I had expected the RSSI output to look like.  However it is not on the same time period nor does it have the same low pulse width and as mentioned the waveform does not look even close.  I am not seeing at any time even with radio off a 1% duty cycle hence the need for low end calibration
PWM channels look fine.  Of course the is for the D4R-II for my measurements.

Have tried on 2 D4R-II units with same waveform results.  Nonetheless it is a usable output.

StewLG, 2015-09-17:
There's now a library that covers this:

https://github.com/diydrones/ardupilot/pull/2770

magicrub, 2015-09-17:
Closed per @StewLG 's comment

closed 2015-09-17 by magicrub