ArduPilot/ardupilot #1828: Anetnna-tracker, standalone.

AndKe, 2015-01-29:
Today, Mission planner does the calculations, and when it crashes, (on a legal EVLOS/BLOS) mission it's hard to restore the connection due to antenna direction.

It would be nice if the tracking APM could be equipped with it's own GPS+Magnetometer , and were tracking fully automatic, just listening in to MAVlink.

rmackay9, 2015-01-29:
The Antenna tracker as described on this wiki page, http://copter.ardupilot.com/wiki/common-antennatracker-introduction/, does just listen to the vehicle's gps position and tracks the vehicle.  It doesn't rely on the mission planner to do the calculations.

There may be some confusion with the antenna tracker feature of the mission planner but in that case this issue should go into the mission planner issues list.

I think I'll close this issue if that's ok.

closed 2015-01-29 by rmackay9

AndKe, 2015-01-29:
ok, I thought it was just like the PWM-output devices MP accepts, and were confused by the lack of GPS on it.   Thank you.  If you know the internal workings of it, how does it determine it's position/yaw ? - (without GPS/Mag).

rmackay9, 2015-01-29:
@AndKe, no troubles, there's a fair bit of confusion between the two types of antenna tracker.

If using the antenna tracker that I posted a wiki link to above, it relies on using an APM2.x or Pixhawk's compass.  So in the case of an APM2.6 you'll need to attach a GPS/compass module or some other external compass.

AndKe, 2015-01-29:
@rmackay9 Thank you. That's basically what I wished to hear, it looked as if it depended on GCS software like the rest :)