ArduPilot/ardupilot #1835: Error when Disarming in Auto Mode

SSUAV, 2015-02-01:
On Disarming while in Auto Mode, i.e after Auto Landing, The error message ""verify_nav:Invalid or no current......" appeares on HUD and Throttle goes to full (GCS failsafe on). Does not happen if switch to FBWA or Manual Modes before disarming.

Arduplane: V3.2.0
MP 1.3.18 build 1.1.5503.19823

magicrub, 2015-02-01:
I have seen this too. I see this when Booting up in auto, without the
transmitter connected and first waypoint is takeoff. Motor spins up when
armed. If you switch to manual then back to auto the behavior is normal.

tridge, 2015-02-03:
thanks for reporting that! I've reproduced it and put a fix into a new beta (3.2.1beta3). It should be built by the build server in an hour or so

closed 2015-02-03 by tridge