ArduPilot/ardupilot #1837: Durning accelcal apm needs to continue to send heartbeat messages

gmorph, 2015-02-02:
In a GCS (MavProxy for example) when you run the command accelcal to calibrate the accelerometers the APM stops sending heartbeat messages and the GCS thinks the link is down which it actually isn't.  Need to update the code to continue to send heartbeat messages during calibration so it doesn't think the link is down.

OXINARF, 2017-06-02:
This is marked for AVR boards which we don't support anymore. Accel calibration method has also changed from when this was opened and I believe it is no longer true that heartbeats are no longer sent.

@gmorph If you think this still happens then please re-open it.

closed 2017-06-02 by OXINARF

khancyr, 2017-06-02:
it is still true for gyro calibration, it is pretty annoying

OXINARF, 2017-06-02:
@khancyr Yes, gyro calibration is different, it is done in one step. It is thought and done as part of init and if the gyros aren't stable it can take up to 30 seconds without doing other tasks. I've never seen that though and it's always fast for me.

WickedShell, 2017-06-02:
@OXINARF share your aircraft during boot, it's really confusing but you can hold it in gyro cal indefinitely, and on PH2 if you don't have an external LED then you don't get a great indication of whats going on which can be annoying.

OXINARF, 2017-06-02:
@WickedShell Indefinitely? How does that happen? Looking at code (https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor.cpp#L1019) I don't see any loop that could stay indefinitely, but maybe I'm missing something.

WickedShell, 2017-06-02:
@OXINARF sorry as long as you are shaking the plane you are holding it in the gyro cal. It's not getting stuck or anything like that, it can just appear it if the user is being silly...

OXINARF, 2017-06-02:
I understood that it was when shaking the plane, but I still don't understand how that holds it in gyro cal - unless Plane is forcing gyro calibrations until gyros are all calibrated (which I would find weird).

WickedShell, 2017-06-02:
Heh, you appear to be right, I've never tried shaking it for more then 15 seconds :P Just tested and it does eventually move on :)