ArduPilot/ardupilot #1848: Check CRC of IO Firmware is correct before arming

gmorph, 2015-02-03:
Under certain scenarios its possible that the programming of the IO Firmware fails.  To mitigate this ArduPilot should check the CRC of the IO firmware to confirm it agrees with the version it was built with.  If there is a mis-match it should fail to ARM and inform the user.

jschall, 2015-02-03:
Who is working on this?

billbonney, 2016-10-03:
To fix this do the following
1. Copy http://firmware.us.ardupilot.org/PX4IO/latest/PX4IO/px4io-v2.bin to the SD Card of the PH
2. Rename it px4io2.bin
3. Put back in PH and power cycle with the safety button held down.

OXINARF, 2017-06-02:
Is this something that is still wanted? At this moment ArduPilot checks CRC and if it doesn't match it tries to update it, but if it fails it simply prints to debug console and plays a tone.

It should be very easy to implement this at this point by failing to continue initialization, but I'm not sure if everyone agrees with that.

WickedShell, 2017-06-02:
I'd like to have a much more clear indicator of failure so either halting the boot, or failing an arming check would be nice. I'd like to keep this open until that is done, as flying with a disagreement in versions could be really bad.

peterbarker, 2017-06-06:
General agreement seemed to be "we should check, someone should really do this!".

WickedShell, 2019-04-05:
Happens in ChibiOS builds as far as I can tell. There are some loop holes with rev2 bootloaders that I think cause you to never pass the check, but the fundamental check is there.

closed 2019-04-05 by WickedShell