ArduPilot/ardupilot #1849: Failsafe in RTL mode

lcasassa, 2015-02-03:
If the plane is in RTL mode, fail safe should not enter to CICLE mode.

This is my fail safe configuration:
THR_FS_VALUE = 1
FS_SHORT_ACTN = 1 with 1.5 seg
FS_LONG_ACTN = 1 with 20 seg

lcasassa, 2015-02-04:
Even more, if the mission has a DO_LAND_START, the plane changes from RTL to AUTO and we could get a failsafe in AUTO that will end on a RTL and restart the DO_LAND_START. So we would have a loop and the plane never landing.

Workaround would be to turn off the radio but on the landing this would be unsafe.

lcasassa, 2015-02-04:
I was playing with the SITL trying to reproduce a crash I had and got to this problem.

tridge, 2015-02-05:
I think we could definitely be smarter about this, though we'll need to think about the exact semantics. I'll add it as a 3.2.2 issue

Naterater, 2018-12-01:
Implemented. See RTL_AUTOLAND parameter: 

[image]

closed 2018-12-01 by Naterater