ArduPilot/ardupilot #1850: FENCE_AUTOENABLE + DO_LAND_START

lcasassa, 2015-02-03:
I am not sure of the APM behavior. Maybe this is not an issue. So please check if I am right or not.

With FENCE_AUTOENABLE = 1, only a LAND will disable the fence.

After a RTL mode, a DO_LAND_START auto mission starts and the way points could be out of the fence. It would be good to have a disable fence command or simply disable it for all the DO_LAND_START mission.

tridge, 2015-02-05:
yes, good point, thanks! I'll add it for 3.2.2

Naterater, 2019-01-04:
Is there still no check for this? I don't recall, but there may be some sanity checks for waypoints outside of fences.

IamPete1, 2021-08-14:
Looks like fence is now disabled for both VTOL landing and forward flight landings. I'm not sure it's really a good idea to disable the fence for the whole of a do land start, that should be inside the fence too surely?

peterbarker, 2024-07-17:
> Looks like fence is now disabled for both VTOL landing and forward flight landings. I'm not sure it's really a good idea to disable the fence for the whole of a do land start, that should be inside the fence too surely?

Agreed.

Closing this now - if people can make a case for disabling all fences when landing, please re-open.

closed 2024-07-17 by peterbarker