ArduPilot/ardupilot #1854: Serial2 is broken

magicrub, 2015-02-04:
Something in Serial2 just broke between v3.2.1alpha2 and v3.2.2alpha1. I can not communicate via MAVLink over serial2 using SERIAL2_PROTOCOL=1 (GCS Mavlink) like I used to.

magicrub, 2015-02-04:
This was found using ArduPlane on pixhawk hardware with Feb 4, 2015 master branch.

rmackay9, 2015-02-05:
@magicrub,
Thanks for the report.  If you set the protocol to MavLink2 it'll work.
Tridge and I have had a discussion and we're going to simplify it so there's just "MAVlink" no "MAVLink1" and "MAVLink2".

magicrub, 2015-02-05:
Hi Randy,

I can't get it to work on SERIAL2_PROTOCOL=2 either. I've tried them all
just in case.

// @Values: 1:GCS Mavlink, 2:GCS Mavlink (2nd), 3:Frsky D-PORT, 4:Frsky
S-PORT, 5:GPS, 6:GPS (2nd), 7:Alexmos Gimbal Serial

Is there anything else I need to be aware of for the new serial manager to
work on serial 2 for a GCS? How did you guys test this?

-TomP

magicrub, 2015-02-06:
After some "buddy testing" with Randy, we were able to duplicate the bug
together but then it resolved itself and neither of us could reproduce
again. At one point Randy had a strange, "old", baud rate for his serial2
and had a thought that perhaps the eeprom value was not loaded correctly.

I just came across this problem with my lidar. I was using the SF02 on
analog pin 14. I would reboot and it would show -1, then again be 14. then
14 then 14 ect but voltage was always read as zero. I loaded Plane v3.2.1
and suddenly the sonar worked, no changes to params. I backdated to the
"broken" firmware which was v3.2.1beta1 (just before the serialManager) and
now the lidar was back to working, still no changes to params. Same issue
with the eeprom not being respected.

-TomP

magicrub, 2015-04-06:
possibly caused by https://github.com/diydrones/ardupilot/pull/2058

rmackay9, 2015-05-13:
It's possible we have a bug somewhere still but the main request of this issue has been completed - the ability to just specify the protocol as MAVLink instead of having to specify MAVLink1,2,etc.

closed 2015-05-13 by rmackay9