ArduPilot/ardupilot #1864: APM Planner v2.0.16-rc3

MJBeaubien, 2015-02-09:
I updated my APM Planner this morning and now when I go to Graphs and try to go to Droneshare APM Planner crashes. I've tried numerous time and crashes every time.

rmackay9, 2015-02-10:
@MJBeaubien, could you raise this in the APM Planner issues list?
https://github.com/diydrones/apm_planner/issues

closed 2015-02-10 by rmackay9

MJBeaubien, 2015-02-10:
Sure.

 

Date: Mon, 9 Feb 2015 17:26:34 -0800

https://github.com/diydrones/apm_planner/issues

MJBeaubien, 2015-02-10:
Done.

 

From: mcbeaubien@hotmail.com
To: reply@reply.github.com
Subject: RE: [ardupilot] APM Planner v2.0.16-rc3 (#1864)
Date: Mon, 9 Feb 2015 17:44:27 -0800

Sure.

 

Date: Mon, 9 Feb 2015 17:26:34 -0800

https://github.com/diydrones/apm_planner/issues