ArduPilot/ardupilot #1877: Plane Landing - disarm after landing

magicrub, 2015-02-14:
After a plane landing is detected a disarm should occur. This can be implemented via a delay param such as LAND_DISARM_DELAY where 0 is disabled and positive values are in seconds before the disarm.

proficnc, 2015-02-14:
Keep in mind petrol aircraft!  A disarm before the motor is off is really bad :)

magicrub, 2015-02-14:
the short term workaround is to disable the feature with a value of 0. However, to properly implement it I suppose we would need to check that throttle is at or near the min. The implementation I was thinking of using was to trigger /w the landing wp which suppresses the throttle anyway.

I'll make sure to reset the timer if we takeoff again (come out of LANDING_APPROACH_FINAL)

magicrub, 2015-02-26:
See pull request 1922
https://github.com/diydrones/ardupilot/pull/1922

closed 2015-04-06 by magicrub