ArduPilot/ardupilot #1878: Abnormaly high Altitude value when using PX4Flow

GunMan-fr, 2015-02-14:
Hi guys :)

I was testing the PX4Flow stuff on the Arducopter master (3.3-dev). Had the following results : 
- When Pixhawk starts up, with PX4Flow_Enable = 1, no particular problem
- After two short seconds, altitude values skyrocket to 20972 m ! 

- Removing the PX4Flow I2C Link and rebooting the Pixhawk fixes the problem.

Some more information :
- My PX4Flow is the one with the sonar included.
- It has been flashed with this firmware : PX4Flow-KLT - Beta-06Dec2014,
- The EKF was enabled (don't know if it's relevant...). 

I will check the PX4Flow module and raw output values in the next few days and post the results here.
Either it is a problem with the altitude estimator, either it's a problem from my PX4Flow. I think that either way it is an issue : faulty values from the PX4Flow should not be taken into account for altitude estimation if they are detected as faulty. 

Thanks for your help ;-)

GunMan-fr, 2015-02-14:
@rmackay9 , tridge told me that you might be interested in this :)

jschall, 2015-02-14:
This is a bug in the user-settable home patches that recently went in.

GunMan-fr, 2015-02-14:
@jschall : Could you be a bit more specific about the "home patches" ? I could not see what's related in the latest commits (arducopter code newbie here :-) but willing to help).

jschall, 2015-02-14:
The block of commits starting with e081b9d1 and ending with 77d8f88e.

git revert e081b9d1^..77d8f88e will probably do the trick.

@jschall <https://github.com/jschall> : Could you be a bit more specific
about the "home patches" ? I could not see what's related in the latest
commits (arducopter code newbie here :-) but willing to help).

rmackay9, 2015-02-14:
Yes, Tridge and I had a quick look at this earlier today.  I'll have another look over the next couple of days.
The issue certainly appeared with the recent do-set-home changes I put in but it may be that those changes simply exposed an existing bug in the EKF or inertial nav lib's get_origin call.
Anyway, it'll get sorted over he next couple of days.

GunMan-fr, 2015-02-14:
Thanks for taking a look at this :) ! I'll test your fix as soon as it's available and post the results here.

rmackay9, 2015-02-15:
Ok, I believe I've found and fixed the issue.  It was an uninitialised Location being returned by the inertial nav's get_origin function.
https://github.com/diydrones/ardupilot/commit/8e75c9580ca454ed6e8906615f277f05d3419a6e

closed 2015-02-15 by rmackay9

GunMan-fr, 2015-02-20:
I confirm this is fixed. Second issue that may be related : GPS FIX is always at 1.00 even if there's 0.00 sats. Is this normal behaviour ?

rmackay9, 2015-02-20:
@GunMan-fr, I think that's normal.  0 is probably when there's no GPS attached at all.

magicrub, 2015-02-20:
I believe GPS Fix values 0,1,2,3 are for Nope, seeking, 2D fix, 3D fix.

GunMan-fr, 2015-02-20:
Oh, allright it makes sense then. To be honest it is a bit counterintuitive in Mission Planner, when in the "Quick" tab it displays 1.00 for GPS FIX. You expect this value to be a boolean and not that "1 = Seeking". Thanks for the answers ;)