ArduPilot/ardupilot #1883: MIS_RESTART and RTL_AUTOLAND

lcasassa, 2015-02-17:
With MIS_RESTART set to 1 will not work with RTL_AUTOLAND set to 1 and with a DO_START_LAND in the mission.

This is because, when it starts the landing sequence it enters AUTO mode and restarts to waypoint #1 instead of continuing the landing.

lcasassa, 2015-02-17:
How about making a new mode like RTL_LAND? We could easily fix the issue where the failsafe kicks in while doing the landing sequence so that it does not restart the RTL/landing sequence (Issue #1849 ).

aldofad, 2020-01-21:
Dear developers, 
has this issue been resolved?

Cheers,
aldo

ClockeNessMnstr, 2020-05-04:
Also struggling with this today on plane. Will need to try and setup rst channel instead but it would be nice not to need an extra channel setup.

ClockeNessMnstr, 2020-05-04:
> Dear developers,
> i see the issue is still open and I confirm it is present in Arduplane 4.0.2, just impacted during this days :)
> 
> Cheers,
> aldo

As a work around you should be able to set an extra channel to reset the auto mission If you have a channel available.

IamPete1, 2021-07-12:
This should have been fixed by https://github.com/ArduPilot/ardupilot/pull/13652
I'm going to close this, if anyone can confirm its still a issue please re-open.

closed 2021-07-12 by IamPete1