ArduPilot/ardupilot #1891: Plane - altitude drift in v3.2.2

magicrub, 2015-02-19:
I'm seeing a lot of altitude drift on my Pixhawk on Plane v3.2.2 which does not happen in v3.2.1alpha (Dec 17). The baro pressure is stable but the AHRS value drifts significantly. This is all while the plane is sitting on the table.

haogodent2, 2015-02-19:
It's the same issue I posted on the user forum below:
http://ardupilot.com/forum/viewtopic.php?f=107&t=11323

magicrub, 2015-02-20:
Does anyone else observe this? It's quite obvious and easy to test and keeps us from using V3.2.x.

tridge, 2015-02-20:
thanks. I've reproduced the issue and I should have a fix this morning

magicrub, 2015-02-20:
eager to alpha test! :)

magicrub, 2015-02-26:
fixed with this commit
https://github.com/diydrones/ardupilot/commit/6781a8d3290d846e9abcb925b7fee0302f5000e2

closed 2015-02-26 by magicrub