ArduPilot/ardupilot #1894: ROI Height discrepancy

conradallen, 2015-02-20:
Hi, guys.  Firstly tank you  for all your hard efforts, very much appreciated by the community as  whole.  

My issue is that the ROI when set seems to be a little too high for the go pro camera to keep the ROI in centre of the frame.  I flew a mission and set the ROI at -10m and it was much better but still had the ROI in the lower half of the frame.  It is the same story in follow me unless the vehicle or person is moving quite quickly to ensure that they are kept in the centre.

I hope i have explained that clearly enough. Perhaps the GPS data fro the follow me could give the correct height for the follow me ROI and an off-set could be worked into normal missions?  You will know better than I how to solve the, if indeed it needs solving and isn't just a quirk we need to embrace.

rmackay9, 2015-02-20:
My guess is that this is actually a setup issue and not a mistake in the firmware.  For example this could happen if the mount wasn't set-up with the correct angle range.

I haven't heard people complain of this and the code should point the camera directly at the target.

conradallen, 2015-02-20:
Thanks Randy, 

I was hoping that was the case.  Where do you suggest I begin looking?  As far as I can tell I have followed all of the video tutorials and wiki pages to the letter.  Any help at all and directio as to where to begin would be gratefully accepted.

Best regards C

Conrad Allen
Conrad.Allen@hotmail.com

amilcarlucas, 2017-09-25:
Please follow the instructions from:
http://ardupilot.org/copter/docs/common-camera-gimbal.html

 and reopen this if the problem presists

closed 2017-09-25 by amilcarlucas