ArduPilot/ardupilot #1896: AP_Limits: Wrong headers included ("GPS.h")

dgrat, 2015-02-21:
I wonder why it is compiling with your buildsystem.
Currently implement a cmake build system and found this.

Best, Daniel

19504643, 2016-02-01:
hello dgrat,
I run into this problem suddenly and I can NOT find where GPS.h is.
did you solve this strange problem?

dgrat, 2016-02-01:
This issue is one year old. I don't think it is still important.

closed 2016-02-01 by dgrat

19504643, 2016-02-02:
except for this GPS.h header error, in AP_Limit_Geofence.h,there comes another error, [GPS *&gps](https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Limits/AP_Limit_Geofence.h)
where is the GPS  defined?

dgrat, 2016-02-02:
It isn't defined, this is why I issued the problem. GPS was a very old class, which is removed already since years. I wrote this problem in the mailing list and in the issue and even made a patch once.

19504643, 2016-02-02:
that is to say,files in the AP_Limits directory are NOT compiled all the way?

dgrat, 2016-02-02:
Yes :D