ArduPilot/ardupilot #1898: Keep track of battery resistance and optionally warn if it's too low before arming

marcmerlin, 2015-02-21:
I had a discussion with Tridge about an expensive (and potentially dangerous) crash with an octocopter where the 6000mAh battery (with about 10 uses) died flat after 4300mAh used.

He mentioned that while the details could be tricky, especially if you swap batteries, the APM could keep track of the battery internal resistance and warn before takeoff if it has increased too much.
This could help detect a battery that is likely to die before its expected capacity.

He suggested I file this bug :)

lthall, 2015-02-22:
Hi Marc,
We are currently adding code that estimates the battery impedance each time you arm and take off. This is being used for another purpose at the moment but we may be able to include the feature you are suggesting.

I don't know how we could calculate battery impedance before take off and be able to abort take off in a safe manner. If you have any suggestions about how battery impedance could be measured without the need for significant current draws (and therefore copter take off), I would welcome your suggestions!!

Do you have any documentation on the characteristics of battery impedance  during failure?

marcmerlin, 2015-02-22:
Hi Leonard, thanks for your answer. I'm not sure how to include Tridge on this, he seemed to have an idea on how to do it when we chatted. Do you know how to Cc him on a github ticket?
I can't give you info from the battery that caused a crash since it ran dry in flight and is now damaged way beyond what it was before takeoff.

But if I had to guess, how about this:
When you arm, it does spin the props at low speed, is that good enough to tell?
If not, could arm spin the props faster until the accelerometers detect the quad is about to take off, and then spin down? That should give a better I to calculate R=U/I .

jschall, 2015-02-22:
I think the solution to this is smart batteries that store battery state
information onboard.

They are already prevalent in commercial and consumer multicopters but have
yet to work their way into the hobbyist market.

lthall, 2015-02-22:
The only thing I can suggest is sounding an alarm if the battery impedance is greater than a user specified value. That would then leave it up to the pilot to decide to land and check the battery.

marcmerlin, 2016-03-29:
@lthall you mentioned a year ago there was code being added to help with this.
Do you know if it happened and whether we should resolve this?

OXINARF, 2017-06-04:
Linked to #6351

lthall, 2018-01-11:
The only practical implementation of the remaining part of this issue is to initiate a failsafe if the battery impedance is over a user defined amount after a predetermined flight time or mAh usage percentage.

Otherwise this is a smart battery option and could be closed.

Naterater, 2019-01-14:
I think this should be closed in favor of smart battery options.

auturgy, 2019-01-15:
Smart batteries aren’t prolific enough yet.

Regards,

James

Evan1010, 2019-01-15:
I've been keeping an eye out for good off-the-shelf modules to make smart batteries from bare batteries - I think once that happens, we can consider the smart battery to be mainstream, as there's less chance that they'll suddenly disappear from the market.
The closest I've seen is this one: https://diydrones.com/profiles/blogs/batmon-pixhawk-compatible-in-flight-battery-monitors-with-led
Which I don't believe monitors cell impedance, but it's a good reference. I got excited because it has partial compatibility with ArduPilot already (its reported cell voltages are not yet supported).
I will admit my qualifier for smart batteries are >6s (So just using a laptop battery chipset is right out), and a usable current capability (>50A) with coulomb counting. Those two make smart batteries for UAVs harder than they seem at first glance . . . I tried designing one myself, and it didn't end well.

So in all, I think it'll be a while before smart batteries are matured, capable, and - as auturgy pointed out - prolific.

peterbarker, 2024-07-17:
I've added a scripting example which will give the desired functionality.

closed 2024-07-21 by peterbarker