ArduPilot/ardupilot #1900: Copter: Autotune not reponding to RC inputs

ronhogwarts, 2015-02-21:
I am having an issue with autotune not responding to RC inputs to reposition, even on full rates, so I have to resort to turning off auto tune, repositioning, and switching it on again to resume. I can provide a normal log file, or provide another with the RC inputs logging enabled if needs be. I am using the latest AC 3.2.1, but was also an issue on 3.2. AC is running on the HK pilot micro (hobbyking APM).

lthall, 2015-02-22:
Hi ronhogwarts,
A log would be great!!

ronhogwarts, 2015-02-22:
I have the "nearly all" log below, showing the RC inputs not being accepted whilst autotuning, I checked it with the APM log file analyser.

Log file:
https://drive.google.com/file/d/0B6QFq1-xwWDSOXBGa1hlMWxCdU0/view

Analyser output:
https://drive.google.com/file/d/0B6QFq1-xwWDSTzloOWplYmpfZFU/view

Is autotune designed to allow pause and resume when it is switched off and on again whilst running? I was forced to do so in this instance due to a light wind.

lthall, 2015-02-22:
That log doesn't show you trying to adjust position while autotune is enabled. I can see that you did not try to reposition the quad in this flight by looking at the rc inputs. We need to see a log where you have not switched autotune off and are attempting to reposition.

ronhogwarts, 2015-02-22:
OK, this took some digging to find all the information, I think this shows the effect. I have the Taranis logs on the left, and the Arducopter log on the right, the scaling is similar. Notice the small excursion to 1460 on the left of the arducopter and the -60 on the left Taranis log at 78 seconds. If you notice on the taranis logs, approximately halfway between the positive and negative excursions at both ends (78 and 104 seconds) there are multiple larger negative excursions at around 89 seconds, at which time the autotune is still on according to SC (switch c, the autotune switch), yet the arducopter logs miss this somehow, and doesn't see a significant input, If you look very closely on the arducopter logs there is an apparent 5us square wave before the alt hold kicks in, which follows the inputs according to the taranis. I'm sorry this effect isn't more pronounced, but I can do another autotune with lots of inputs to highlight this further.
[image]

rmackay9, 2015-02-23:
This is really a support question and should go to the APM Forum and then if it's confirmed to be a bug should be posted here.  Sorry to be annoying, it's just that if we don' enforce this our git hub enhancement list turns into a support form.

closed 2015-02-23 by rmackay9

lthall, 2015-02-23:
Hi ronhogwarts,
This looks like you have set the taranis to not output when you have enabled autotune. If you want to start a support request I am happy to talk to you there but the two plots above are pretty convincing to me.

ronhogwarts, 2015-02-23:
Yes, on closer inspection that's the issue, an issue with the mixes on the taranis, good spot.