ArduPilot/ardupilot #1901: Copter: Autotune gains far too low for mini quads

ronhogwarts, 2015-02-21:
The gains that auto tune provides for pitch and roll are far too low on the latest AC 3.2.1,or 3.2 for my 550g 250 Pro frame. I believe this is a common problem, and I have followed the advice on the forums to no avail, please see my thread here for further details:
http://ardupilot.com/forum/viewtopic.php?f=100&t=11079&p=29527#p29527

From my understanding the autotune procedure will be altered significantly for 3.3 with yaw gains, and be faster, so perhaps this has been fixed there, if anyone can confirm?

lthall, 2015-02-22:
Hi ronhogwarts,
I replied to your thread and asked for logs so I can see what is happening with your auto tune.

While it would be great to make auto tune more robust, when it doesn't have a problem it results in pids that are slightly too high, not as you say, too low.

When autotune reaches minimum rate D of 0.002 (or near that), it is a sign that there is an offset CG, cross coupling between axis, high IMU rate vibrations, alt hold or one of the other axis is near oscillation, poor noise on the IMU accel causing a bad alt hold, or a number of other issues.

While we will continue to make auto tune more robust and intelligent, it does require a half reasonable starting point and mechanically sound airframe. The bigger the airframe, the easier this tends to be. It is for this reason I do most of my development on an an underpowered QAV250 and will be adding a stupidly over powered 250 soon.

I think this issue can e closed.

ronhogwarts, 2015-02-22:
I'm happy to continue this discussion with you on the forum and mark this as closed, it may a particular case here, but once your overpowered 250 comes to light it'll be interesting if you encounter similar issues or not.

closed 2018-01-11 by lthall