ArduPilot/ardupilot #1902: APM::Copter 3.1.x / 3.2: going from STAB to RTL/failsafe causes heavy shaking

marcmerlin, 2015-02-21:
On a 3DR octocopter (X8) every time I hit failsafe, 

easier to show than to explain:
http://youtu.be/d_f7-pGPbos?t=1m13s (OSD)
http://youtu.be/UBm7jdLaxiM?t=0m55s (gopro)
(watch for 15 seconds, no need to watch more)

I did hit RTL due to the altitude fence, that's ok, but the way the X8 shakes and fights my input when that happens is worrisome, including the 45 degree or so attitude it seems to hit every time this kind of failsafe is met and the mode change happens.

Here's another very clear example of undesirable behaviour, from flying in STAB to hitting RTL due to 2/3rd of the battery mAh used:
http://youtu.be/UBm7jdLaxiM?t=5m40s (watch 20sec or so)

Any idea what's going on?
log from that flight: http://marc.merlins.org/tmp/9410eaf3-9820-4911-a71e-4444fb71abf8.bin

omersiar, 2015-02-24:
When I look at your log file i can tell you that there is similarity in graph lines between COMPASS(MAG) and MOTOR OUTPUT (RCOUT). When APM hits RTL it tries to understand its heading and in your situation it leads shaking problem. Someone who expert on this can help you more

marcmerlin, 2015-02-24:
Thanks for having a look. Do you have someone you know you can forward this to?
I have no idea who to contact :)

omersiar, 2015-02-24:
You can use diydrones' forum i think

rmackay9, 2015-03-19:
We have another issue raised to make entering RTL more gentle: https://github.com/diydrones/ardupilot/issues/1971

I had a quick look at the logs.  It looks like the battery voltage is getting very low after only 5min of flight.  Without voltage scaling this leads to the effective gains (i.e. attitude response is weak) being much lower at the end of the flight which leads to some very scary overshoot.

AC3.3 does voltage scaling (although it will likely be disabled by default) which may help.
Getting a bigger battery may also help.
reducing the ANGLE_MAX to 3000 (30deg will make the vehicle generally more docile).
The issue mentioned above to make entering RTL smoother will also help but that may not make AC3.3.

closed 2015-03-19 by rmackay9

marcmerlin, 2015-03-19:
Thanks rmackay. The battery being too low was indeed a failed battery that went stone dead after 70% of its rated capacity, so that was of course the reason for the crash during landing, but indeed the heavy shaking and the copter fighting my input (trying to go opposite from me and shaking) was not linked the battery, it happened earlier in the flight and on other flights.
I hope issue 1971 will indeed fix this.

As for ANGLE_MAX I think I had 45% because I wanted to be able to fly a higher horizontal speed and also noticed that the RTL/LOIT just had very slow horizontal speed, so I changed other parameters to make sure the AP wasn't limiting itself in unnecessary ways (before I changed those, it would take so long to come back on its own that it could just run out of batteries before coming back too slowly)

rmackay9, 2015-03-20:
@marcmerlin,
Normally reducing the angle-max will reduce the acceleration but maybe not the top speed.  I haven't personally tested this but I've heard the vehicle's top speed is achieved around 22deg (depending upon the power-to-weight ratio with a more powerful copter able to lean over more).  It's really the vehicle's acceleration which is reduced.
To increase the RTL speed it's the WPNAV_SPEED that should be increased to 1000 or 1500 (15m/s).  for loiter it's a separate parameter.. WPNAV_LOIT_SPEED I think.

marcmerlin, 2015-03-20:
@rmackay9 thanks for the tips.
Indeed I did also set:
WPNAV_LOIT_SPEED 2000 (20m/s)
WPNAV_SPEED 2000 (20m/s) = 72km/h (20m/s max)
WPNAV_SPEED_UP 450 (33s for 150m)
WPNAV_SPEED_DN 600 (33s for 200m)
WPNAV_ACCEL -> 500
WPNAV_ACCEL_Z -> 300

Good to know that I didn't need to set the max angle to 45 degrees, thanks.

R-Lefebvre, 2015-03-20:
I don't see the connection between 22 degrees and maximum speed.  This is a highly complicated topic.  It depends on a lot of factors.