ArduPilot/ardupilot #1903: Pulselight_LRF power saving

Naish88, 2015-02-22:
Implement power saving mode for Pulselight Lidar Lite. This could be done it this way. 
 If (EKF estimated altitude - SRTM elevation ) is more than ( LRF max range x 1.5 ) then, send sleep command thought I2Cbus. 
If no SRTM data is loaded, then don't use the sleep function 
~100mha in continuous mode is a lot of power if you are looking for endurance on an electric plane

magicrub, 2015-05-06:
This has been implemented in plane via PR https://github.com/diydrones/ardupilot/pull/2055 in a slightly different way and for a different reason. It only works for PWM mode and it uses the enable pin on the lidar connector via the RNGFND_STOP_PIN feature. The new feature will power down the module until the plane starts a landing approach. This is done not for power saving. Instead if is to avoid/limit GPS jamming by the lidar causing degraded GPS performance and inability, or much longer than that normal, 3D lock behavior.

Naterater, 2019-01-07:
I'm assuming that the http://ardupilot.org/plane/docs/parameters.html#rngfnd-stop-pin-rangefinder-stop-pin parameter doesn't accomplish this through the requested i2c logic, is that right?

peterbarker, 2026-11-15:
#2055 was never merged.

#33712 added similar support for TFMiniPlus, so there is infrastructure support for it.