ArduPilot/ardupilot #1906: Plane: Always departs loiter WP towards next WP

iskess, 2015-02-22:
I like to use the LOITER_TURNS command to create a short holding pattern to allow the airplane to stabilize its altitude before continuing with the approach to auto land. The problem is the plane is not consistent with the direction it departs the Loiter, and will often depart in the wrong direction creating a very irregular and unpredictable approach path. I don't see any reason why the code shouldn't require the departure from loiter to always be in the direction of the next waypoint. I realize this might not give you exactly the number of turns you requested, depending on where you entered the Loiter, but that variability is preferable over flight path uncertainty.

tridge, 2015-02-23:
yes, I agree. To make it consistent we'd probably make it break out of the loiter the next time a vector directly out the front of the aircraft crosses the next WP target.

iskess, 2015-02-23:
That would be perfect! Thank you.

magicrub, 2015-06-10:
fixed with https://github.com/diydrones/ardupilot/pull/1981

iskess, 2016-02-28:
Still having problems with exiting loiter on the correct heading.
[image]

[2016-02-25 13-31-52.zip](https://github.com/diydrones/ardupilot/files/149776/2016-02-25.13-31-52.zip)

magicrub, 2016-02-28:
I agree this should be fix/changed. Meanwhile, have you tried
loiter-to-alt? It has a heading to next waypoint optional requirement which
is exactly what you want.

iskess, 2016-02-28:
Tom, can you reopen this issue?
I didn't use loiter to alt in this case because I use the two loiter turns to buy me time to get mentally prepared for the approach. I could have used loiter time, but I wanted the loiter to end heading the correct direction, and I'm not sure if loiter time will exit immediately when the time expires.

magicrub, 2016-02-28:
reopened. There is a new command loiter-to-alt but I agree that all loiter commands should have this behavior.

iskess, 2016-02-28:
Strange, it was working great for a while there, now it's broken again.

magicrub, 2016-02-28:
it was luck then of when it started, it's never had that feature.

WickedShell, 2016-02-29:
Shouldn't that be optional? IE if my radius is large a faster flight path
would actually be to just leave the loiter regardless so that the plane can
pull it's max bank rate to get there?

Although I agree that this is very reasonable default behavior to always
exit towards the next WP, as that is what people probably expect
intuitively.

iskess, 2016-02-29:
Sorry, I'm not following. Could you try to explain again? Why would I be loitering if I want the fastest flight path?

WickedShell, 2016-02-29:
Loiter to alt is usually where I use it, although there is also an argument
for loiter time. . IE I tookoff executed takeoff logic to 50 m but want to
ensure I'm safely clear of all obstacles enroute to the mission so I then
have a loiter to alt to take me to 125m then I want to go to the next point
in the mission.

iskess, 2016-02-29:
My example was loiter turns, but I do do your point with loiter time, or loiter to alt.

magicrub, 2016-02-29:
there's an option in loiter-to-alt, I'm sure it can be extended to the
rest.

magicrub, 2016-03-01:
pending PR https://github.com/diydrones/ardupilot/pull/3694

OXINARF, 2017-01-07:
@iskess @magicrub Was this fixed with #3694?

magicrub, 2017-01-07:
Yes.

closed 2017-01-13 by iskess