ArduPilot/ardupilot #1907: Increase accuracy of co-ordinates

gmorph, 2015-02-23:
At the moment co-ordinates (including waypoints) are communicated through MAVLink as float's.  This means their accuracy is limited to about 1 meter (it varies a little depending on where you are in the world).  We need to convert them to int's which should then provide cm accuracy.

Note: I SWEAR I had already raised this but I couldn't find - must be going mad.

LorenzMeier, 2015-02-28:
@gmorph That's an usage issue. Just use the right messages ;-).
http://mavlink.org/messages/common#COMMAND_INT
And see MAV_PROTOCOL_CAPABILITY for the right flag to indicate the AP supports it.

gmorph, 2015-03-02:
Yep that's right.  Just adding it to the list of tasks that need a code change.

njoubert, 2015-05-15:
I'm working on this at the moment :)

rmackay9, 2015-05-15:
and for mission items we should accept these I guess:
http://mavlink.org/messages/common#MISSION_ITEM_INT

LorenzMeier, 2015-05-15:
If possible I would appreciate if the switch to mission_item_int also does away with the magic waypoint for home handling. At the very, very least the home position should have a distinct action (e.g. DO_SET_HOME) so it doesn't confuse autopilots or GCS implementing standard MAVLink.

The reason this makes sense is that the MISSION_ITEM_INT support is encoded by the autopilot capabilities message and flags, and so there would be an easy way to tell that this handling is different.

If you opt for using a distinct action for home even that would not be required. I'm happy to set up a call how the cleanup could be done without disrupting deployed systems and without introducing complexity or stability issues.

njoubert, 2015-05-15:
+1 @ng-Laser

ng-Laser, 2015-05-16:
Looking through ArduPilot, we dont see any place where the capabilities variable actually gets set to anything other than 0. Where should we set this flag to show that we now support COMMAND_INT?

njoubert, 2015-05-16:
Hi Lorentz!

Yea that all sounds great! Here's how I want to proceed with this:

For the moment I need accurate ROI point for my current research paper due in 3 weeks, so we just added this capability (me and @ng-Laser). Once the paper is in we will then switch waypoints and mission items over as well. Before we do that we'll set up a call with you. Sounds good?

m11s, 2016-03-02:
@gmorph @LorenzMeier
MISSION_ITEM__INT support has been added to Mission Planner and is pending pull request in AP.
https://github.com/diydrones/ardupilot/pull/3688`

OXINARF, 2017-06-02:
MISSION_ITEM_INT support has been added for some time now. COMMAND_INT is supported but only for one or two commands.

@gmorph Do you feel this still deserves to be open?

amilcarlucas, 2018-10-19:
Waypoints now support MISSION_ITEM_INT

WickedShell, 2018-10-19:
`COMMAND_INT` needs to be handled the same way `MISSION_ITEM_INT` is or we are shooting a lot of vehicles/use cases in the foot. I'm going to reopen this so we have something to track that.

It's a harder task then the `MISSION_ITEM_INT` stuff, because we already have some support for `COMMAND_INT` and there are some implementation differences that have crept in. We need to try and eliminate those as well.

peterbarker, 2018-10-20:
https://github.com/ArduPilot/ardupilot/pull/8839 will help.

rmackay9, 2023-09-13:
Yeah, I think we can close this one probably.

peterbarker, 2024-05-23:
Closing this now as done, surprisingly.  We accept all commands as command_long and translate them to command_int, as Grant predicted.

I wonder if we *could* have fixed the home-in-mission-slot-0 problem as Lorenz suggested.  I doubt it, considering the want `MISSION_ITEM_INT`/`MISSION_ITEM`/`MISSION_REQUEST` work.   We probably need a capability bit indicating whether misison-slot-0 is home in ArduPilot, have that in place for several years and have GCSs understand it - *then* we could look at removing it from slot-0.  Even if we chose not to remove it from slot-0 the capa bit would allow GCSs to know how to work with the autopilot.

closed 2024-05-23 by peterbarker