ArduPilot/ardupilot #1908: Plane: Request: Loiter until target altitude is reached

iskess, 2015-02-23:
I read this request before but could not find it here on Github. The ability to loiter until a selected altitude is reached would be very useful when operating an areas of steep terrain. 
I have tried to use CONDITION_CHANGE_ALTITUDE but I can't get it to work. Even if I could get it working, it requires several commands and it is not very intuitive. I found these instructions by Doug Weibel:
(http://diydrones.com/forum/topics/flight-planner-question?overrideMobileRedirect=1)
"You can also do it with a series of 3 commands.  I believe it would go like this for example...
mission command 9 - Waypoint (before loiter)
mission command 10 - Condition Change Alt
mission command 11 - Do Jump (to command 13)
mission command 12 - Loiter unlimited
mission command 13 - Waypoint (after loiter and altitude change)"

I would love to see a new command called LOITER_ALT_CHANGE.

tridge, 2015-02-23:
yes, a single mission item makes sense for this, and wouldn't be hard to do

davidbuzz, 2015-02-23:
Wouldn't it make more sense to just do the mission in 3d space in the first place?  Ie don't consider a way point reached until the altitude indicated inside the way point is actually reached.   Fly - over and fly - by way points can be similar acting  in the z direction too.

iskess, 2015-02-23:
@davidbuzz,
Interesting thought, could you describe it better? What happens to the plane laterally if it doesn't meet the vertical requirement? Does it automatically enter a loiter at the WP? In full scale aviation these are called "crossing restrictions" and they are built into the departure and arrival procedures. If the pilot is not in a position to abide by a crossing restriction the software will alert the pilot, but no lateral deviation will occur without ATC approval because that could cause a traffic conflict. I see the same issue here with allowing a loiter that isn't planned for. It could result in an unexpected flight path. 
However if the Loiter_Alt_Change idea is used, the pilot is prepared for a loiter. The autopilot will try to meet the crossing restriction prior to entering the loiter, and if it succeeds, it should continue ahead with no loiter.

davidbuzz, 2015-02-23:
If someone currently sets a waypoint with a 100m altitude ... it either means they want the aircraft to do everything it can to be at 100m (+/- some margin relating to the size of yhe WP) by the time it arrives at the WP... or it should.

magicrub, 2015-02-23:
Buzz, I disagree. There is currently no expectation of achieving an
altitude of a waypoint and that allows for a clean and expected flightplan.
Perhaps what you're looking for is a flag in the waypoint to enforce
altitude

magicrub, 2015-05-02:
released in APv3.3.0 via https://github.com/diydrones/ardupilot/pull/1981

closed 2015-05-02 by magicrub