ArduPilot/ardupilot #1913: Is there any documentation for SIM_* parameters?

nickponline, 2015-02-26:
Can't see any reference to them in the source and I'm wondering what they control?

```
SIM_ACCEL_FAIL   0.000000
SIM_ACC_RND      0.000000
SIM_ASPD_RND     0.500000
SIM_BARO_RND     0.000000
SIM_BATT_VOLTAGE 12.600000
SIM_DRIFT_SPEED  0.000000
SIM_DRIFT_TIME   0.000000
SIM_ENGINE_MUL   1.000000
SIM_GPS_BYTELOSS 0.000000
SIM_GPS_DELAY    2.000000
SIM_GPS_DISABLE  0.000000
SIM_GPS_GLITCH_X 0.000000
SIM_GPS_GLITCH_Y 0.000000
SIM_GPS_GLITCH_Z 0.000000
SIM_GPS_HZ       5.000000
SIM_GPS_NUMSATS  10.000000
SIM_GPS_TYPE     1.000000
SIM_GYR_RND      0.000000
SIM_MAG_ERROR    0.000000
SIM_MAG_RND      0.000000
SIM_SERVO_RATE   0.000000
SIM_WIND_DIR     180.000000
SIM_WIND_SPD     0.000000
SIM_WIND_TURB    0.000000
```

naktinis, 2016-02-26:
Happy birthday, issue!:)

Naterater, 2019-01-07:
I think this is more of a discussion/question rather than an issue. I think you might receive more response on discuss.ardupilot.org. If you believe this is a documentation issue, maybe that issue could be raised in the Wiki repository. Closing for now.

closed 2019-01-07 by Naterater

L00rdD, 2023-01-05:
https://ardupilot.org/dev/docs/sitl-parameters.html

nickponline, 2023-01-05:
Stop the clock! 7 years, 10 months, 10 days. Can close.