ArduPilot/ardupilot #1915: Copter: limited vertical speed in althold when thr_max<1000

cecca00, 2015-02-26:
I'm currently using some blheli ESCs in closed loop mode, to have correct rpms I need to scale output pwm between 1000 and 1700us, so I set thr_max to 700.
When I give full stick in althold or loiter mode the copter doesn't climb at full desired vertical speed.
My pilot_velz_max is 8m/s, but as you can see in the picture it stops at 2m/s.
[image]
Arducopter 3.2.1 on Pixhawk

rmackay9, 2015-02-27:
@cecca00, can you post a log?  If it's confirmed to be a bug we can sort it out.
Generally these requests should go into the APM Forum first by the way.  We don't want the github issues list (which is for enhancements and confirmed bugs) turning into a support form.

cecca00, 2015-02-27:
The graph above is pretty clear and easy to reproduce by everyone even with normal open loop ESCs.
Full stick down the target speed is 8m/s as set, full stick up the target speed is 2m/s.
I don't think this is the intended behaviour.

R-Lefebvre, 2015-02-27:
Randy, pretty sure this is real, and is a long standing problem.

cecca00, 2015-02-28:
Yes it's real. An easy workaround is leaving THR_MAX at 1000 and change TCRV.
Maybe THR_MAX just needs to be removed, MOT_TCRV_MAXPCT do the same job and do it well.
I also think a better throttle curve should be implemented, a 3 points broken line couldn't be really helpful in linearizing any thrust curve, mine with closed loop escs is something x^2.
But that's another story.

rmackay9, 2015-03-01:
@cecca00, there's an improved throttle curve going into AC3.3.  It's just going through a final review before going into master.
I'd really like to see a dataflash log of this.  I hear two respectable people saying it's real but I'd really like to see a log to help find the cause.

cecca00, 2015-03-01:
No problem, here are 2 logs, log 130 as a reference with thr_max 1000 and tcrv_maxpct 66 (tcrv_midpct 33 to have a straight line), log 131 with thr_max 660 and no throttle curve. In both cases thr_mid set to hover with stick in the middle.
Another problem I noticed with decreased thr_max is the copter becomes untuned and oscillates a bit.

http://www.mediafire.com/download/tuxkm1h6f4a4vkm/LOGS_130_131.7z

lthall, 2018-01-11:
This is old and I don't believe this has been a problem for years.

closed 2018-01-11 by lthall