ArduPilot/ardupilot #1916: Password

abvgeej, 2015-02-26:
A simple password auth may make it times harder to hijack the drone.
Encryption is a separate matter with long discussions on DIYDrones, but a password might be a very useful and easy quickfix.

CodeChief, 2015-02-26:
Password is an unsecured token which is easily discovered. With today's know-how all we need is a public key of the trusted ground stations stored on the SD card of the APM/Pixhawk and a one time (arming) CPU usage to authenticate and generate a token used for the short battery life we have. With an occasional signature to prevent capture/replay, nobody will ever discover the key in that short time and CPU/math/battery is minimized and pretty good security achieved.

abvgeej, 2015-02-26:
@CodeChief I know that it is a very unsecure token which can be easily sniffed, but right now, one can hijack a drone with functionality provided by MissionPlanner, which is in no way a good thing.
Another thing, is that it might be hard to find a password in the field if all the settings were preset, and no password is transmitted.
Yet, if your implementation wont take long discussions on DIYDrones, it is times better.
I proposed this as a quickfix, before anybody is there to put some proper protection in place, and password auth should be easy to implement

magicrub, 2015-02-26:
In a TCP session like telnet this gives some level of protection because
the port rejects others from connecting. However, for serial links this
does not help at all.

CodeChief, 2015-02-26:
The link type is regardless, if you're blocking the signal it'll failsafe or crash. Given a signal we are always talking some kind of protocol and security can be applied. Why be negative? The solution is straight forward. Maybe I implement a branch of this out of interest. Drones are cheap enough to be expensive toys now so it's just a matter of time before stupid kids on the block start hijacking hobbyists for fun. The same thing happened with bluetooth phones in the early days after the mobile boom. It's a real threat there should be no discussion just solutions put forwards.

magicrub, 2015-02-26:
related post
<https://groups.google.com/forum/#!searchin/drones-discuss/aes/drones-discuss/MfHFsWTFXKs/kgV4t4H9h1cJ>

abvgeej, 2015-02-27:
@CodeChief Unfortunately, having +8 channels is still black magic to APM:Copter.
If you control the telemetry, in my theory, it is dead easy to send the drone to your location without the owner being able to do anything, and it will take me nothing to make a POC device out of my 50$ telemetry. And its 50$ for a official, not enchanced, expencive telemetry.
The main radio can always be jammed or exploited, but its a different unprotected side of the drone, which does not allow you to flash your custom firmware into the drone or the radio like the telemetry.

WickedShell, 2017-02-10:
MAVLink2 offers packet signing, which gives you authorization requirements. It doesn't obscure any data, but it does fulfill the issue here. There is a separate issue for tracking encryption.

closed 2017-02-10 by WickedShell