ArduPilot/ardupilot #1917: Annoying bright, flashing LED

abvgeej, 2015-02-26:
"I need to see you, and you want to burn my eyes" i said once again putting the drone on the ground.
There should be a way to set maximum brightness for both armed and disarmed states, as well as having a less offencive "PX4-style" LED flash for the disarmed state.

magicrub, 2015-02-26:
An easy, "low-tech", solution is to cut the wire and put a resistor in
series. That will lower the brightness. In your case, I would suggest a
turn-pot to dial in a brightness that suits your needs. A more complicated
solution is to convert the GPIO into a user-adjustable PWM duty which may
not be very easy depending on the particular pin and available software
timers.

abvgeej, 2015-02-26:
@magicrub But isnt it possible from software? PX4 does it fully from software.

magicrub, 2015-02-26:
Yes, that is the more complicated solution I mentioned

abvgeej, 2015-02-26:
@magicrub Everything can be done given enough resistors, time and space, but i would prefer to have a software solution for this.

CodeChief, 2015-02-26:
This falls under the same category as the mute annoying beeps request which has appeared here before and whilst my eyes are maybe not so good as yours I know what you mean. A couple of small comfort & convenience features in the APM firmware would be welcome. Let's call them LED scale and beeper disable. My example is for the many hours you have to work on your drone at home to get things right, which is plain annoying when "field scale" beeps and LEDS fire off every other minute.

CodeChief, 2015-02-26:
By the way I resorted to installing a micro DIP switch on the beeper to solve my issue but would still prefer a software (firmware) solution and LED brightness dimming would be good for me too.

abvgeej, 2015-02-26:
@CodeChief I just pulled the cable of the buzzer, but i wont dare doing this with the led.
I am almost used to this LED burning my eyes out as a pre-flight procedure, but i would still like it to be under control. Possibly with a extra knob to control its brightness.

pepevalbe, 2016-03-30:
This should be closed, right?

CodeChief, 2016-03-30:
Well we never got an answer really. I would like to know... 1) Is it technically possible (should be if the LED is controlled by a PWM/RGB value), is the beeper possible to adjust/stagger volume. 2) Is anyone in the team interested in implementing that, presumably straightforward then.

pepevalbe, 2016-03-31:
@CodeChief  you can reduce LED brightness: http://ardupilot.org/plane/docs/parameters.html#ntf-led-bright-led-brightness

WickedShell, 2017-11-03:
As was pointed out `NTF_LED_BRIGHT` controls this for all states (and allows turning it off as well), so closing the issue.

closed 2017-11-03 by WickedShell