ArduPilot/ardupilot #1919: Hand-pause

abvgeej, 2015-02-26:
It would be very convinient with a light drone, if you could catch it in your hand, and pause the rotors for some time, and then arm it in your hand immediately by moving it on its nose, and back.
I understand the risk of injuries, but i am experienced with adjusting the equipment on a in-air drone.

rmackay9, 2015-02-27:
This is a real edge case and sounds dangerous.  I don't think we will implement this.  If someone does implement this then we can consider pulling it in but until then I'm going to close this issue.

closed 2015-02-27 by rmackay9

abvgeej, 2015-02-27:
Does not look like a edge case to me. There are quite alot of situations where flat ground is unavailable, and/or you want just a quick pause, to for ex. move it through a area unpossible to fly through, wait until the filmed object is going to stand back on snow, or just redo some of the equipment on the drone.

I would be happy even just for a method to arm it by tilting it forward and back again, or any other gesture you find good.
I would really find it useful.

rmackay9, 2015-02-27:
@abvgeej, arming/disarming while the vehicle is in the user's hand sounds very dangerous to me.  I certainly bend to the community's wishes though so if you can gather up a couple more people to support the idea...

abvgeej, 2015-02-27:
So bad i have no friends in the community :(

magicrub, 2015-02-27:
Would this apply to boat takeoffs?

rmackay9, 2015-02-27:
The boat take-offs we will definitely support and handlign that is more about enhancing the gyro calibration so it can happen even when the vehicle is moving.  There's a separate issues somewhere on this issues list for this.

@abvgeej, sorry to be a kill-joy but adding a feature that somehow implicitly implies that holding a copter in your hand is "ok" sounds like a bad idea.  I cringe when I see videos of people holding, pulling, pushing their copters around while they're flying because it's dangerous behaviour and generally there's no good reason, it's just kinda fun like playing with matches.

magicrub, 2015-02-27:
But but but pulling my quad around while in loiter is the coolest demo of
them all!

rmackay9, 2015-02-27:
@magicrub, yes, it has replaced the act of putting your transmitter on the ground as the sign of true awesomeness.

R-Lefebvre, 2015-02-27:
Agreed, sorry, but this sounds like a really terrible idea.  I don't support this feature at all.

abvgeej, 2015-03-02:
This would bring us a step forward to being as easy to use as a Micro-multirotor, where the fun is about just being able to quickly deploy it, from any surface, and taking off from a hand is one of the greatest thing in them.
You can already arm and take off from a hand, but the user should grow a third hand, or have his feet in the cold.
I am talking about a implementation where once the quad is tilted forward and back again in a straight position, it would arm and _SLOWLY_ gain the throttle set by the transmitter.
I understand that we are not in the business of "Hubsans", but of machines which can kill with easy if misused, but "When we want to burn a pillow, we want to burn a pillow."

hotelzululima, 2015-03-09:
Do folk stick their hands into a running waring blender?(yes)

NO way am i getting close to multicopter but a nano in flight.. and even the CX-10 is perfectly capable of cutting skin around the nose and eyes(hard experience)..

I concur with Rob.. this is a REALLY TERRIBLE idea even for copters with blade shrouds...(unless the shrouds are complete)... and even then... MURPHY...

         hzl

abvgeej, 2015-03-09:
Is it just that everyone thinks that the fact that the functionality is available will motivate people to use it, or I am the only person who feels fine and confident with the copter "touching his head with the landing gear", and adjusting the camera "in air"?

rmackay9, 2015-03-10:
I'm happy to revisit this request as an optional feature in arducopter if there's a copter which is small enough that you can stick your fingers in the props without getting hurt and it's got a flight controller board on it that can run arducopter.

R-Lefebvre, 2015-03-10:
Agreed. Sort of.  Bebop drone class, alright. But then it's hard to prevent it being abused by people with larger machines.

abvgeej, 2015-03-10:
I am to use it on my "large machine". The question is whether it will get it here, or I need to start learning ardupilot code. (A nice opportunity to do so, and probably the only choice for me :smile:)
After all, its not a request to add a bug. Its a feature which is useful if used properly.

@rmackay9 Yeah, I am, and probably the most of us are still waiting for such a drone :smile: